ngardiner/TWCManager #533: 1.3 // Can not save Tesla token or car's location -error

danielsan1, 2023-10-01:
Hey @MikeBishop  @ngardiner @tjikkun 

I thought, maybe I messed up the docker installation as I'm not used to docker. So I tried the the manual installation but the error stays the same :( Can you help me with that?

**When I try to save the Tesla token as asked for after installation:**

Exception occurred during processing of request from ('192.168.188.23', 52903)
Traceback (most recent call last):
  File "/usr/lib/python3.9/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 158, in __init__
    BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
  File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.9/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
    method()
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 985, in do_POST
    res = master.getModuleByName("TeslaAPI").saveApiToken(url)
  File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1058, in saveApiToken
    params = json.loads(req.text)
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

**And the same when I try to reset my home location:**

Exception occurred during processing of request from ('192.168.188.23', 52862)
Traceback (most recent call last):
  File "/usr/lib/python3.9/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 158, in __init__
    BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
  File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.9/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
    method()
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 941, in do_POST
    self.process_home_location()
  File "/home/pi/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 1141, in process_home_location
    del master.settings["homeLat"]
KeyError: 'homeLat'

**What can I do? Everything else runs smoothly...**

Thank you all in advance!
I really hope to get 1.3 running at some point :)

danielsan1, 2023-11-11:
Hey @ngardiner and @MikeBishop - can you please have a quick look into this? I would really appreciate your help, as also my 1.2.6 installation out of the sudden stoped working and I'm now completely without TWCManager :(

(I tried the manual and Docker installation and I get this same error in both cases!)

BIG thanks in advance!

MikeBishop, 2023-11-13:
Without digging further into the code, it looks like the `settings.json` might be corrupted -- the first error looks like it's trying to load the file and re-save it with the token, while the second is trying to remove(?) the location from it.  Try renaming your `settings.json` and letting TWCManager create a fresh one.  (That will reset your tokens and some schedules, which you'll want to re-add.)

danielsan1, 2023-11-15:
Thank you very much @MikeBishop ! I’ll try that out on the weekend 🤞

dehsgr, 2023-11-15:
@MikeBishop deleting old settings.json doesn't work. Having the same issue here from within docker:
```
Exception occurred during processing of request from ('192.168.2.101', 64048)
Traceback (most recent call last):
  File "/usr/lib/python3.9/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/src/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 158, in __init__
    BaseHTTPRequestHandler.__init__(self, *args, **kwargs)
  File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.9/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
    method()
  File "/usr/src/TWCManager/lib/TWCManager/Control/HTTPControl.py", line 985, in do_POST
    res = master.getModuleByName("TeslaAPI").saveApiToken(url)
  File "/usr/src/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1058, in saveApiToken
    params = json.loads(req.text)
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```
Thinking this might me related to Tesla API changes. Isn't it?

ngardiner, 2023-11-15:
I strongly suspect so. I will get a release out within 24 hours to address it, appreciate the importance

dehsgr, 2023-11-18:
@danielsan1 yeah that seems to be a workaround. 👌🏼

dehsgr, 2023-11-18:
@ngardiner will you make the new release available via docker too?

danielsan1, 2023-12-18:
Hey @ngardiner any news on the Tesla API issue? I updated to 1.3.1 (web interface update still do not work for me) but I still hat to use the workaround and put the two keys manually in via the settings page... (and unfortunately the system asks me every few days again about the keys... that was not the case as "copy the auth.tesla.com..... URL in" still worked)

Thanks for your help!

ngardiner, 2023-12-19:
> Hey @ngardiner any news on the Tesla API issue? I updated to 1.3.1 (web interface update still do not work for me) but I still hat to use the workaround and put the two keys manually in via the [...]

I think this is due to the location API issue. Just to flag that I believe we may be as close as two weeks away (going on Tesla's start of 2024 timeline) from Tesla changing the APIs all over again. The current API that we use (the command API) is about to be replaced by the Tesla Fleet API:

https://developer.tesla.com/docs/fleet-api#membership-tiers

I am not entirely sure how this is going to go. I would not call it entirely favourable but we will need to find a way forward as we have with all the previous non developer friendly Tesla API changes. In particular, we will now be subject to these:

[image]

On the bright side I guess, we're a lot _less_ subject to these than solutions which limit the charge rate via API, so we'll probably be fine, given we only use the API for starting and stopping charging. I'm just a little jaded about Tesla API changes by this point so...

ngardiner, 2023-12-19:
Anyway the point of the above is to brace for a few more changes, but hopefully given this is an actual official documented API, these might... might be the last changes for a while, and if so I think we'd all be grateful for that.

danielsan1, 2023-12-20:
Thank you very much for the detailed explaination! So let's keep or fingers crossed! :)

MikeBishop, 2023-12-20:
Yeah, it's a little unclear exactly how the new API is going to work, still.  There's basically no affordance for open source / self-hosted projects like this; it assumes that each app that's going to use it has publicly accessible web servers.  What's likely to happen, I'd think, is that each person will have to register for a developer API key and then have themselves as their single customer.  Setting that up may be onerous for most people, unfortunately.

I'd hope to see a way for individuals to get access to their own vehicles using a simpler flow, and leave the complex dance for apps that need access to vehicles across many accounts.  We'll see if that happens; I'm bracing for all my Tesla-related apps to simultaneously break in January.

dtiefnig, 2023-12-21:
> On the bright side I guess, we're a lot _less_ subject to these than solutions which limit the charge rate via API, so we'll probably be fine, given we only use the API for starting and stopping charging. I'm just a little jaded about Tesla API changes by this point so...

Might be worth reworking the location tracking part, as if I understood correctly (and I'm really not sure about that) this queries the API once per minute. As stated in the 2023.38+ update, using [Fleet Telemetry](https://github.com/teslamotors/fleet-telemetry) might be worth a try. Didn't look into this yet though.

Curious how the membership tiers will look like. Having a shared partner level account may be troublesome, as a single deployment could mess up API access for everyone sharing the account.

Let's hope there's an "owner" tier which allows easy access for car owners. From the experience with Tesla so far I'm not quite confident about this. Calling the thing "FleetAPI" doesn't improve my expectations, but then OTOH it also includes "energy devices" so they can't be really serious about the naming in general. :)

MikeBishop, 2023-12-21:
Location won't call any more often than once per minute, but doesn't actually call that often most of the time. Typically hourly, or when a car is plugged/unplugged to check if it just arrived/left. And of course, if you're connected to Teslamate, it's just listening to the location from there and never fetches the location directly.

dehsgr, 2023-12-27:
> I just tried […] to add the access token and the refresh token manually and it seems to work! […]

This workaround doesn‘t work for me since a few days. :(

danielsan1, 2023-12-27:
Same here :-/ Seems like another change on Teslas side…

How will it affect TWCManager? Solar charging will still work but it won’t stop automatically but continue until full with 6A, even the solar output is lower than 6A?

MikeBishop, 2023-12-29:
They've said that...

> We're having some internal discussion on the best way forward for self-hosted applications, including discussions about providing an endpoint for obtaining personal-use Fleet API tokens without a developer account or dedicated domain. We don't want to force self-hosted apps to migrate to a centralized model, but we also want to avoid inadvertantly incentivizing centralized apps to solicit user passwords or personal-use Fleet API tokens.

That suggests that while we may need to switch to the new API, there will be a path forward. Of course, if they don't come up with their solution for self-hosting before they sunset the old API, we'll be out of luck at least temporarily.

Flitzer42, 2024-02-18:
Hi all, 
thanks for all your efforts and the great work.
Have the same issues as danielsan1 when I install TWCManager on a blank system from scratch. No chance to get it running.

To add here:
When Start/Stop charging did not work any more, I updated my existing installation (1.3.0) running for more than a year with "git pull" "git checkout main" "make install" manually on a pi.
With that, getting the tokes with "Tesla Login" and "past URL here" still works fine. 
Also starting and stopping charging works fine, even the car sleeps.
But after a day, the token expires and I have to refresh them again.

My takeaways:
1.) In principle it works, but current installation / make is missing a part that is necessary (and was there with Buster).
2.) Tokens are expiring within a few hours (max. one day). 

More investigations:
My running (old) installation runs on "buster" with only python 3.7. There is no python 3.9.
Setting up a pi on old buster and installing TWCManager "main" let me retrieve and successfully enter the token.

Any Ideas what's wrong?