ngardiner/TWCManager #534: TWC clean install - error when trying access via HTTP

MiRoCZ, 2023-11-05:
Hello @ngardiner, my old TWC installation (SD card) failed, so I have to go for new / fresh install. Unfortunatelly when trying all ways (Manual/Docker, Stable/Main version) always getting some error when I run TWC (only Docker 1.2.6 stable version is OK, but can´t be upgraded to 1.3.0 which is latest stable version for Manual installation, but not available as docker 1.3.0 version - am I correct?).

For example docker install of main version completed successfully:

[image]

But when accessing via browser, no page is displayed and followinf error occurs in console:
[image]

Same issue with web access for manual installations.

Installed on new hardware:
rPi Zero 2W, 8GB Industrial micro SDcard, latest rPi Imager v1.8.1, Raspberry OI IS (Legacy) Debian Bullseye with security updates and desktop environment - released 2023-05-03 

I strictly followed installation instructions (updated version for manual installation).

Please advice. Thank you.

Samiam789, 2023-11-06:
I had a similar error when trying to access the webpage. There was an error with my HASS module for polling my solar. When I fixed that the error disappeared.

MiRoCZ, 2023-11-06:
For new clean manual installation I´m getting this error:

[image]

Any advice please ?

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.

closed 2026-10-26 by MikeBishop