ngardiner/TWCManager #537: Auth failure, start/stop charging failure

TitusJohannes, 2023-11-24:
Dear community,
yesterday, my two Tesla on the my two TWCs stopped reacting on the start/stop commands from my TWCManager 1.3.0 (running since months on docker image).
I deleted the docker images and repositories and installed the "latest" via docker. However, the "latest" docker installation according to the description given shows again v.1.3.0. 
Thus I decided to stop and delete these docker containers and images and installed "manually" according to the description the 1.3.1 version. That worked and is now at 1.3.1, but the authentication failed: The "home" page state to follow step 1 and 2 and I pasted the successfully generated token to the commandline. The consequence is a http://192.168.0.108:8080/teslaAccount/saveToken site opened with "connection lost". If I come back the site http://192.168.0.108:8080 again, the token is required and the connection to my cars is lost.

Do you have any idea what is wrong here?

Thanks for any help!

dehsgr, 2023-11-24:
See #533 …

TitusJohannes, 2023-11-25:
Thanks dehsgr! The authentication is working now normally!

However, the start/stop charging does still not work. In my opinion, the problem is the reading of the cars GPS data (MS and MX). When I set my current car gps data as my home GPS point, the "homeLon" and "homeLat" is set to "10000" which is interpreted as the northpole. When I set my home gps point back to the correct values in the settings.json, it is shown correctly in the map in the settings menue. However, if the cars gps data is not interpreted correctly, and this is what I guess, the car will not start charging even when at home...
Thanks again for your help!

r3pTiLe1860, 2024-01-21:
Any chance to have this fixed? Can‘t use TWCManager currently because it‘s continously charging below 6 ampere.

Today i had to refresh token - no problem, but it still doesn‘t work to control car. Also no SOC is displayed and so on.

Got this error message:

```
an 21 17:07:57 raspizero python3[5028]:   File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 256, in background_tasks_thread
Jan 21 17:07:57 raspizero python3[5028]:     carapi.applyChargeLimit(limit=task["limit"])
Jan 21 17:07:57 raspizero python3[5028]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 878, in applyChargeLimit
Jan 21 17:07:57 raspizero python3[5028]:     if not vehicle.update_charge() or not vehicle.update_location():
Jan 21 17:07:57 raspizero python3[5028]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1436, in update_charge
Jan 21 17:07:57 raspizero python3[5028]:     return self.update_vehicle_data()
Jan 21 17:07:57 raspizero python3[5028]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1420, in update_vehicle_data
Jan 21 17:07:57 raspizero python3[5028]:     drive = response["drive_state"]
Jan 21 17:07:57 raspizero python3[5028]: KeyError: 'drive_state'
Jan 21 17:07:57 raspizero python3[5028]: , occurred when processing background task
```

TitusJohannes, 2024-01-21:
Hi,

I did not get rid of the problem.

I had to change the tokens every some days but the automatic refresh of the token failed in my system.

This was the reason why I switched to TeslaConnect in the Loxberry and I organize the start and stop charge management with TeslaConnect. Here the auth flow works well. The TWC Manager I use for setting the charge current without using the functionality of the Start-Stop Charging. Therefore I can neglect the non working auth flow. On the other hand, you can send all commands and read out all vehicle data with TeslaConnect. There are many opportunities to integrate the Tesla in smart homes with this tool.

 

Yours sincerely,
Titus

MikeBishop, 2024-01-22:
For some reason, perhaps the API changes, the JSON object being returned does not contain the attributes we're expecting. What surprises me is that it fails at that line, rather than finding something in the response which indicates an error.

r3pTiLe1860, 2024-01-22:
Can i supply any more that helps solving it? Or do you have the same result?

What i also noticed is, that my car doesn‘t sleep anymore while twcmanager is running since 2 months. So i had to stop it to have my car go to sleep again (otherwise high vampire drain - Model S with still MCU 1)

Looking forward to have again the state that correctly starts/stops charging and also let my car sleep :-)

closed 2026-04-14 by ngardiner