ngardiner/TWCManager #552: External controll

unichargepay, 2024-01-29:
Hello.
First of all, I'm sorry if this not a kinf of an issue.

I'm new to the project and need some help.
I managed to get it running on my Raspberry pi.

Is it possible to externally (by API, MQTT or something) enable or disable charging?
For example, I want to ultilize a rfid card reader, that will enable charging at full current, using a simple python script
to communicate with the TWCmanager.

Is this possible?

Thank you, very much.

MikeBishop, 2024-02-08:
There is an HTTP-based API. If TWCManager is already controlling charging, you can override the offered charge rate with the [chargeNow](https://github.com/ngardiner/TWCManager/blob/main/docs/modules/control_HTTP_API/chargeNow.md) command, giving a current and duration.  You can cancel it early with [cancelChargeNow](https://github.com/ngardiner/TWCManager/blob/main/docs/modules/control_HTTP_API/cancelChargeNow.md).

unichargepay, 2024-02-16:
Hello @MikeBishop 

Thank you for your reply.
I did some testing with Nissan eLeaf today.

FIrst thing is that "Solar Surplus" is always active, which i do not want, i don't have solar installation.
I searched a bit and found this:

'''
        "restrictions":{
          "Track Green Energy":{
            "match":[""],
            "condition":["false"],
            "value":[""]

'''

Which was only way, hot to get the "Do Not Charge" button lit.

Second thing.
Even if setting the TWC to "Do not Charge", the car started charging after plugin it in.

I would like to block any kind of charging, unless I tell the charger to do so.

Here are my settings and config,

[settings.json](https://github.com/ngardiner/TWCManager/files/14314038/settings.json)
[config.json](https://github.com/ngardiner/TWCManager/files/14314039/config.json)

Thank you.

unichargepay, 2024-02-28:
@MikeBishop 
Would you know, how to prevent the TWC from charging after connecting the the car?
I mean to be able to charge car only by pressing charge now from the ui, or via api.

I thought about using the offset, but I'm not sure about that?

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.

closed 2026-10-26 by MikeBishop