ngardiner/TWCManager #555: Unable to start/access web interface after update.

alexeiw123, 2024-02-05:
I've just done `git pull` and `make install` to try to resolve the latest tesla API update issues. I can no longer access the twcmanager interface.

```
~/TWCManager $ sudo -u twcmanager python -m TWCManager
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/TWCManager/TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
  File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 636, in <module>
    master.loadSettings()
  File "/home/pi/TWCManager/lib/TWCManager/TWCMaster.py", line 774, in loadSettings
    carapi.setCarApiBearerToken(self.settings.get("carApiBearerToken", ""))
AttributeError: 'NoneType' object has no attribute 'setCarApiBearerToken'
```

RichieB2B, 2024-02-05:
It looks like you cannot load `TeslaAPI.py` so `carapi` stays `None`. Try this:
```
pip3 install jet
```
This is a new requirement since the patch.

alexeiw123, 2024-02-05:
> It looks like you cannot load `TeslaAPI.py` so `carapi` stays `None`. Try this:
> 
> ```
> pip3 install jet
> ```
> 
> This is a new requirement since the patch.

Have done that, switched back to main and run `make install` again.

Still getting the same or similar error.

```
pi@TWCManager:~/TWCManager $ sudo -u twcmanager python -m TWCManager
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/TWCManager/TWCManager.py", line 34, in <module>
    import TWCManager.TWCManager
  File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 636, in <module>
    master.loadSettings()
  File "/home/pi/TWCManager/lib/TWCManager/TWCMaster.py", line 774, in loadSettings
    carapi.setCarApiBearerToken(self.settings.get("carApiBearerToken", ""))
AttributeError: 'NoneType' object has no attribute 'setCarApiBearerToken'
```

RichieB2B, 2024-02-05:
Sorry I (autocorrect) made a typo there. The missing module is `jwt`:
```
pip3 install jwt
```

alexeiw123, 2024-02-05:
> Sorry I made a mistake there. The missing module is `PyJWT`:
Ah, not to worry. I wonder what 'Jet' was? well, I've uninstalled that now.

> ```
> pip3 install PyJWT
> ```

OK, now we're getting somewhere!
on main branch and now can launch TWCManager, I haven't yet tested Tesla API.

This is odd though...
[image]
I've never noticed this on the main branch before?

alexeiw123, 2024-02-05:
> Sorry I made a mistake there. The missing module is `PyJWT`:

Not to worry, I have uninstalled 'jet' whatever that was!

> ```
> pip3 install PyJWT
> ```

OK I've installed PyJWT now and done `git checkout main` and `make install again`
Something odd happened where it booted in to v1.3.0, I don't understand why. In any case I re-did my steps and the main branch installed v1.3.1 and I'm getting the same errors again.

```
pi@TWCManager:~/TWCManager $ sudo service twcmanager restart
^[[Api@TWCManager:~/TWCManagejournalctl -f
-- Logs begin at Mon 2024-02-05 20:06:03 AEDT. --
Feb 06 08:10:59 TWCManager systemd[1]: twcmanager.service: Service RestartSec=100ms expired, scheduling restart.
Feb 06 08:10:59 TWCManager systemd[1]: twcmanager.service: Scheduled restart job, restart counter is at 5.
Feb 06 08:10:59 TWCManager systemd[1]: Stopped Tesla Wall Charger Manager.
Feb 06 08:10:59 TWCManager systemd[1]: twcmanager.service: Start request repeated too quickly.
Feb 06 08:10:59 TWCManager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Feb 06 08:10:59 TWCManager systemd[1]: Failed to start Tesla Wall Charger Manager.
Feb 06 08:11:48 TWCManager sudo[7512]:       pi : TTY=pts/0 ; PWD=/home/pi/TWCManager ; USER=root ; COMMAND=/usr/sbin/service twcmanager restart
Feb 06 08:11:48 TWCManager sudo[7512]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Feb 06 08:11:48 TWCManager systemd[1]: Started Tesla Wall Charger Manager.
Feb 06 08:11:48 TWCManager sudo[7512]: pam_unix(sudo:session): session closed for user root
Feb 06 08:11:55 TWCManager python3[7518]: Traceback (most recent call last):
Feb 06 08:11:55 TWCManager python3[7518]:   File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
Feb 06 08:11:55 TWCManager python3[7518]:     "__main__", mod_spec)
Feb 06 08:11:55 TWCManager python3[7518]:   File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
Feb 06 08:11:55 TWCManager python3[7518]:     exec(code, run_globals)
Feb 06 08:11:55 TWCManager python3[7518]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager/TWCManager.py", line 636, in <module>
Feb 06 08:11:55 TWCManager python3[7518]:     master.loadSettings()
Feb 06 08:11:55 TWCManager python3[7518]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager/TWCMaster.py", line 774, in loadSettings
Feb 06 08:11:55 TWCManager python3[7518]:     carapi.setCarApiBearerToken(self.settings.get("carApiBearerToken", ""))
Feb 06 08:11:55 TWCManager python3[7518]: AttributeError: 'NoneType' object has no attribute 'setCarApiBearerToken'
Feb 06 08:11:55 TWCManager systemd[1]: twcmanager.service: Main process exited, code=exited, status=1/FAILURE
Feb 06 08:11:55 TWCManager systemd[1]: twcmanager.service: Failed with result 'exit-code'.
Feb 06 08:11:55 TWCManager systemd[1]: twcmanager.service: Service RestartSec=100ms expired, scheduling restart.
Feb 06 08:11:56 TWCManager systemd[1]: twcmanager.service: Scheduled restart job, restart counter is at 1.
Feb 06 08:11:56 TWCManager systemd[1]: Stopped Tesla Wall Charger Manager.
```

RichieB2B, 2024-02-05:
It is still failing to load `TeslaAPI.py`. The reason will be in the TWCManager `logfile`. See the `FileManager` directive in your `config.json` for the `logfile` location.

alexeiw123, 2024-02-05:
```
2024-02-06 09:40:22,777 - ⛽ MManager  20 Limiting charging to 12.70A - 10.63A = 2.08A.
2024-02-06 09:40:22,779 - ⛽ MManager  20 Charge when above 2A (minAmpsPerTWC).
2024-02-06 09:40:24,458 - ⛽ MSlave    20 SHB 2679: 00 00.00/00.00A 0000 0000  M: 09 00.00/16.00A 0000 0000
2024-02-06 09:40:24,461 - ⛽ MSlave    20 Slave power for TWCID 2679, status: 0
2024-02-06 09:41:03,360 - ⛽ MManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:41:07,862 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:41:08,949 - ⛽ SManager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 09:41:14,050 - ⛽ SManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:41:31,496 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:41:31,903 - ⛽ MManager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 09:41:37,264 - ⛽ MManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:41:39,218 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:41:39,664 - ⛽ MManager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 09:41:44,570 - ⛽ MManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:41:46,375 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:41:46,702 - ⛽ MManager 240 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 09:41:52,047 - ⛽ MManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:41:53,953 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:41:54,296 - ⛽ MManager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 09:48:21,652 - ⛽ SManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:48:23,503 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:48:23,784 - ⛽ MManager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 09:48:29,023 - ⛽ MManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:48:30,817 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:48:31,086 - ⛽ SManager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 09:48:36,076 - ⛽ SManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:48:37,913 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:48:38,196 - ⛽ SManager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 09:48:43,489 - ⛽ MManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:48:45,335 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:48:45,604 - ⛽ MManager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 09:48:50,608 - ⛽ MManager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 09:48:52,411 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 09:48:52,678 - ⛽ MManager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
```
looks like i'm missing scipy and jwt? whatever they are. is pyjwt different to jwt?

alexeiw123, 2024-02-06:
Unfortunately still the same after doing those pip3 installations.

```
2024-02-06 11:49:15,372 - ⛽ Manager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 11:54:07,229 - ⛽ Manager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 11:54:11,174 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 11:54:11,463 - ⛽ Manager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
2024-02-06 11:54:31,340 - ⛽ Manager  40 ImportError: No module named 'jwt', when importing Vehicle.TeslaAPI, not using Vehicle.TeslaAPI
2024-02-06 11:54:33,256 - 🎮 HTTP     20 Serving at port: 8080
2024-02-06 11:54:33,549 - ⛽ Manager  40 ImportError: No module named 'scipy', when importing EMS.P1Monitor, not using EMS.P1Monitor
```

The scipy install took a very long time to complete too. pi zero W working overdrive!

alexeiw123, 2024-02-06:
pip list is as follows
```
pi@TWCManager:~/TWCManager $ pip3 list
Package            Version
------------------ ---------
asn1crypto         0.24.0
attrs              21.4.0
build              0.7.0
certifi            2018.8.24
cffi               1.15.1
chardet            3.0.4
commentjson        0.9.0
cryptography       42.0.2
entrypoints        0.3
formatizer         0.1.1
future             0.18.2
growattServer      1.2.0
idna               2.6
importlib-metadata 4.11.2
Jinja2             2.11.2
jsonschema         3.2.0
jwt                1.3.1
keyring            17.1.1
keyrings.alt       3.1.1
lark-parser        0.7.8
MarkupSafe         2.0.0rc1
numpy              1.21.6
ocpp               0.14.0
packaging          21.3
paho-mqtt          1.5.1
pep517             0.12.0
pip                18.1
psycopg2           2.9.3
pycparser          2.21
pycrypto           2.6.1
PyGObject          3.30.4
PyJWT              2.8.0
pymodbus           2.5.3
pyModbusTCP        0.1.10
PyMySQL            1.0.2
pyparsing          3.0.7
pyrsistent         0.18.1
pyserial           3.5
python-apt         1.8.4.3
pyxdg              0.25
PyYAML             6.0
requests           2.25.1
scipy              1.7.3
SecretStorage      2.3.1
sentry-sdk         1.5.6
setuptools         68.0.0
six                1.16.0
solaredge-modbus   0.7.0
ssh-import-id      5.7
sysv-ipc           1.1.0
termcolor          1.1.0
tomli              2.0.1
TWCManager         1.3.1
typing-extensions  4.1.1
urllib3            1.24.1
websockets         10.2
wheel              0.32.3
ww                 0.2.1
zipp               3.7.0
```

RichieB2B, 2024-02-06:
It looks like your TWCManager runs in a different environment (venv) than your `pip3 install` and `jwt` is still missing in it. You can apply #556 (manually is fine) which adds `PyJWT` to `requirements.txt` and `setup.py`.

RichieB2B, 2024-02-06:
It looks like your TWCManager runs in a different environment (venv) than your `pip3 install` and `jwt` is still missing in it. You can apply #556 (manually is fine) which adds `PyJWT` to `requirements.txt` and `setup.py`.

Alternatively you can try installing the Raspbian packaged version:
```
apt install python3-jwt
```

alexeiw123, 2024-02-06:
> can apply #556 (manually is fine)

wasn't sure how to do this...

> Alternatively you can try installing the Raspbian packaged version:

This worked a treat! Up and running. Not asking for tokens either so that's a good sign.

PR #556 (2024-02-06): Add jwt to requirements
Hopefully fixes #555
closed 2024-02-06 by ngardiner