ngardiner/TWCManager #557: Clean install error - TWC 1.3.1 stable version - urllib3 (2.2.0) or chardet (4.0.0) doesn't match a supported version!

MiRoCZ, 2024-02-05:
Hello,

I´m repeatedly getting following error on the clean manual install of TWC 1.3.1 - stable version.
I strictly followed installation instructions - https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md - no errors during installation. Installation completed successfully. 

Error message:
/usr/lib/python3/dist-packages/requests/__init__.py:87: RequestsDependencyWarning: urllib3 (2.2.0) or chardet (4.0.0) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "

[image]

Installed on new hardware:
rPi Zero 2W, 8GB Industrial micro SDcard, latest rPi Imager v1.8.5, Raspberry PI OS (Legacy, 32 bit) Debian Bullseye with security updates and desktop environment - released 2023-12-05

I´d like kindly to ask for help.

MiRoCZ, 2024-02-18:
Dear Nathan @ngardiner, @RichieB2B, community,

may I kindly ask for help with this pending issue please?

There is an issue in stable version instalation. 

Please see (clean manual installation) log below and suggest how to resolve or fix instalation.

Thank you very much.

[image]

[image]

RichieB2B, 2024-02-18:
I just installed 1.3.1 in a clean Debian bullseye docker image and it does not have the issue you describe. I suspect you have a conflict between Python libraries installed by Debian and those installed by TWCManager using `pip3`. To fix this you can try to remove the Debian versions using:

```
apt remove python3-urllib3 python3-requests python3-chardet
```
Then rerun `make install`. 

Also, at this stage you want to install the `main` branch since 1.3.1 is most likely not working correctly anymore because of changes in the Tesla API.

MiRoCZ, 2024-02-18:
> I just installed 1.3.1 in a clean Debian bullseye docker image and it does not have the issue you describe. I suspect you have a conflict between Python libraries installed by Debian and those [...]

Hello @RichieB2B  , thank you very much for your response. I have tried, but get this result:
[image]

Anyway, as you mention you doesn´t not have this issue with new clean instalation, can you please refer to updated instalation instructions ? 
I strictly followed following installations instructions and get the error:

Manual: https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md
Docker: https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Docker.md (not updated for 1.3.1)

I apologize for my ignorance as I´m not very familiar with programing, but was previously able to install and configure TWCManager with HomeAssistant and used it successfully for more than year. 

Would you be please willing to help me (over chat or Zoom/Teams) and get functional TWC Manager ? 
I´m ready to pay for it as I´m already desperate :-). THANK YOU !

MiRoCZ, 2024-02-18:
> You should always run `apt` as root:
> 
> ```
> sudo apt remove python3-urllib3 python3-requests python3-chardet
> ```

Sorry @RichieB2B , I might have figured that out. The uninstallation was successful and the error message is no longer displayed. TWC Manager does start up, but unfortunately reports other errors when trying to save API key ...

Could I ask for the step by step procedure on how you installed the docker version, which I can simply replicate and get functional TWC manager ?

P.S: I didn´t used Tesla API key until now and had configured TWC Manager to stop charging by stopping responding to slaves.

RichieB2B, 2024-02-18:
If you don't use the Tesla API then 1.3.1 (or main) should still work for you. What errors are you seeing now?

MiRoCZ, 2024-02-18:
> If you don't use the Tesla API then 1.3.1 (or main) should still work for you. What errors are you seeing now?

Overall, it's acting weirder than how it acted before.

Cannot set the non-scheduled charge rate - the A value is completely missing and cannot be selected. The slider shows 0A and does not allow to change it as it usually did.

[image]
[image]
[image]

Even though I believe I have config.json configured correctly, TWC Manager does not read data from Home Assistant or send data to MQTT (only see 1 attempt (time stamp without valid data) in MQTT Explorer, just after 1st. run of clean instalation and then no more MQTT attempts)

Main edited variables in config.json:
wiringMaxAmpsAllTWCs": 16,
wiringMaxAmpsPerTWC": 16,
minAmpsPerTWC": 6,
numberOfPhases": 3,

HAAS configured and enabled as source, correct IP, valid apiKey
MQQT configured and enabled for status export

I´m pretty lost now and had no clue where is the issue.

RichieB2B, 2024-02-19:
You need to look for clues in the `logfile`, the default location is `/etc/twcmanager/log/`. To make this more verbose I have increased my `logLevel`, you can do this with:
```
"logLevel": 13,
```
Hopefully the issue(s) will stick out. You can also upload the `logfile` here.

RichieB2B, 2024-02-19:
The [installation instructions](https://github.com/ngardiner/TWCManager/blob/main/docs/Software_Manual.md) are still correct, there is no need to change them. There is also no need to start over to fix a permission error. 

You are trying to start TWCManager as the user `pi` without using `sudo`. The correct way to start it is:
```
sudo -u twcmanager python -m TWCManager
```
If you still can't find the logfile, lookup the setting for `path` under `logging` in your `config.json`. Your screenshot shows the content of `~/TWCManager/etc/twcmanager` which is probably `/home/pi/TWCManager/etc/twcmanager` but you should be looking in `/etc/twcmanager` instead:
```
ls -la /etc/twcmanager
```

revoneor, 2025-02-14:
I also had issues with docker image 1.3.x and HASS, Python 3.10
I resorted to source build 1.2.4 and pyenv and venv python 3.7 there the HASS worked with the same config.
the url looks the same in the hass.py, but somehow it doesn't come through. 
It is sensitive to the python version it seems. With 3.10 a curl with same url and apikey works.

RichieB2B, 2025-02-14:
@revoneor How are you using a venv and docker image at the same time? What commands are you using? What are the errors? What doesn't work for you?

It is probably best to open a new issue.

revoneor, 2025-02-17:
> [@revoneor](https://github.com/revoneor) How are you using a venv and docker image at the same time? What commands are you using? What are the errors? What doesn't work for you?
> 
> It is probably best to open a new issue.

Well I am not the specialist so I'd like to have confirmation , I just repeated pyenv , docker compose latest and python 3.10 means no reporting to hass, the other python3.7 does work. I will try out later pythons.

RichieB2B, 2025-02-17:
@revoneor When using `docker-compose` you are running a prebuilt docker image (check your `docker-compose.yml` to see which version) and the python version that comes with it. When using `pyenv` you run TWCManager locally. You cannot combine the two methods. Please open a new issue/discussion. This one is about a specific problem with from a year ago.

ngardiner, 2026-04-26:
Fixed in 6a65a63