ngardiner/TWCManager #560: MQTT - Unsupported callback API version: version 2.0 added a callback_api_version

MiRoCZ, 2024-02-20:
Hello @RichieB2B,

I finally moved forward with clean manual instalation of the main branch, but getting following error when MQTT enabled:

[image]

I found this article - https://stackoverflow.com/questions/77984857/paho-mqtt-unsupported-callback-api-version-error

Is the error caused on my side or there is a need to modify instalation towards new 2.0 version as mentioned by comment in the article - "[Release 2.0.0](https://github.com/eclipse/paho.mqtt.python/releases/tag/v2.0.0) of the Paho Python MQTT includes breaking changes; this means that code written for v1.x will not work without some (minimal) modifications. As [v2.0.0](https://github.com/eclipse/paho.mqtt.python/releases/tag/v2.0.0) was only released a few days ago (11th Feb 2024) most examples"

Thank you for your time and response. 
Have a nice day.

RichieB2B, 2024-02-20:
We will fix MQTT in TWCManager, but for now you can downgrade your MQTT python library manually using:
```
pip3 install -U paho-mqtt<2.0.0
```

MiRoCZ, 2024-02-20:
> We will fix MQTT in TWCManager, but for now you can downgrade your MQTT python library manually using:
> 
> ```
> pip3 install -U 'paho-mqtt<2.0.0'
> ```
MQTT python dowgraded to 1.6.1, but error still pending:

[image]

Also tried `sudo make install` with same result.

MiRoCZ, 2024-02-20:
> We will fix MQTT in TWCManager, but for now you can downgrade your MQTT python library manually using:
> 
> ```
> pip3 install -U 'paho-mqtt<2.0.0'
> ```

Ok, thank you. I run `sudo pip3 install -U 'paho-mqtt<2.0.0'` and MQTT is OK now.