ngardiner/TWCManager #563: Actual status of the new Tesla API integration - for noobs

danielsan1, 2024-03-02:
Hi all,

Again, many thanks for your efforts on the TWCM project!

I really tried to follow the development activities here but it reads quite complex and I’m still not sure what to do with the new Tesla API now.

Can you please help with a short (beginner-friendly :) update if/how we can use the new Tesla API with TWCM 1.3.1? Or can/would you recommend a not-so-difficult workaround? Or would an upgrade to the developer version 1.3.2 help?

Thank you very much in advance!

Daniel

Flitzer42, 2024-03-03:
Hi @danielsan1 ,
I have the same issues as you already wrote in October.
Installing 1.3.1 on a raspberry with the latest OS (bullseye) fails.
Installing on a raspberry with buster (legacy, comes with Python 3.7) works partially. With this, I can at least enter the credential and start / stop charging works fine. But as soon as the the token expires, it stops working. The refresh does not work.
When TWCManager attempts to refresh, the refresh token gets deleted from the settings.json.
I really hope that @MikeBishop or @ngardiner will find a solution.
BTW: I am using OpenWB in parallel and they managed to get it running with the new API.
Klaus

MikeBishop, 2024-03-05:
That PR was merged on 1/23, so after 1.3.1.  Until a new release is cut, you'd need to be on the dev branch to use the new API.  However, the tokens I'm currently using are from the legacy API and continuing to work. If you're using the dev branch and still failing with legacy tokens, that's worth investigating. Hopefully @ngardiner will cut a release soon to include those changes.

Getting tokens for the new API is a bit involved; my attempt at making it as user-friendly as possible is [here](https://github.com/MikeBishop/fleetapi-tokens), if/when it turns out we all need to go that route.

alexeiw123, 2024-03-05:
> That PR was merged on 1/23, so after 1.3.1. Until a new release is cut, you'd need to be on the dev branch to use the new API. However, the tokens I'm currently using are from the legacy API and continuing to work. If you're using the dev branch and still failing with legacy tokens, that's worth investigating. Hopefully @ngardiner will cut a release soon to include those changes.
> 
> Getting tokens for the new API is a bit involved; my attempt at making it as user-friendly as possible is [here](https://github.com/MikeBishop/fleetapi-tokens), if/when it turns out we all need to go that route. Given that Tesla hasn't fully disabled the legacy API yet, I'm hoping they have a selfhost user plan that's better than that.

I'm also having trouble with the legacy API tokens on the dev branch. Sometimes I can get my API and refresh token to 'stick' but it doesn't seem to last. I've changed my configuration to stop responding to slaves and use twc control only for the time being. Interesting to me is that my teslamate connection with legacy API is still humming along just fine.

I see there is a way to leverage teslamate tokens from TWC manager (https://github.com/ngardiner/TWCManager/blob/main/docs/modules/Vehicle_TeslaMate.md) but I've not been able to figure my way through this, given my teslamate is running in a docker and I full flat when the instructions stop!

ngardiner, 2024-03-05:
>I see there is a way to leverage teslamate tokens from TWC manager (https://github.com/ngardiner/TWCManager/blob/main/docs/modules/Vehicle_TeslaMate.md) but I've not been able to figure my way through this, given my teslamate is running in a docker and I full flat when the instructions stop!

Sadly this functionality broke when symmetric encryption was added to TeslaMate to stop people from doing exactly what we were doing.

I haven't removed it because it is entirely possible for us to decrypt the tokens with the encryption key supplied to TM, the problem is time as the encryption approach isn't documented anywhere except the code itself, so it requires reading through the elixir code to work out how it is implemented and to reverse it.

alexeiw123, 2024-03-05:
> Sadly this functionality broke when symmetric encryption was added to TeslaMate to stop people from doing exactly what we were doing.
>
> I haven't removed it because it is entirely possible for us to [...]

Ah, well I'm glad I didn't invest too much time in this. Could it be worth an issue/PR over on the teslamate git now that there's a new development team, to see if teslamate could add support for third party apps to leverage it's connection to the tesla API?

ngardiner, 2024-03-06:
I've been watching this discussion here: https://github.com/teslamate-org/teslamate/discussions/3379
Honestly, having read the code just now it seems trivial, but I was hoping someone else would get to it before me. I'll have a go at it tonight.

alexeiw123, 2024-03-06:
> I've been watching this discussion here: [teslamate-org/teslamate#3379](https://github.com/teslamate-org/teslamate/discussions/3379) Honestly, having read the code just now it seems trivial, but I was hoping someone else would get to it before me. I'll have a go at it tonight.

That's great. If you have luck in that I'd be happy to help you build instructions for a docker install of teslamate. I have no idea how to access the files within the teslamate install inside a container/volume so it would need to be a fairly guided walkthrough.

alexeiw123, 2024-04-01:
Would love a noobs instruction for where to configure the fleet API. I've got the public key hosted using my domain I just don't know how to configure TWCManager to point to it. I can't get the old API to work anymore.

MikeBishop, 2026-10-26:
v1.4.0 adds a Fleet API login flow in the web UI (#639) and shows each Tesla token's status on the Settings page (#692); setup is described in `docs/modules/Vehicle_TeslaAPI.md`. Could you check whether that answers your question?

alexeiw123, 2026-10-26:
@MikeBishop great to see active development continuing here. Instructions in `Vehicle_TeslaAPI.md` look clear and comprehensive but I've since retired my Gen 2 charger and am using different charging hardware, so won't be able to test this.

MikeBishop, 2026-10-27:
Great news! Once #726 merges, you can use TWCManager with no TWC at all! It can do everything over BLE or the API. I'd love for you to try that out, since that's exactly what you've moved to. In the meantime, I'll close this issue.

closed 2026-10-27 by MikeBishop