ngardiner/TWCManager #564: Car API not available - can't stop charging

GMerg, 2024-03-06:
Hello,
Im having some issues for weeks now with trying to stop charging.
In log I see the following:

```
Mar 06 19:05:53 TWCManager python3[471]: 19:05:53 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/REMOVED <Response [401]>
Mar 06 19:05:53 TWCManager python3[471]: 19:05:53 🚗 TeslaAPI 20 ERROR: Can't access vehicle status for XYZ.  Will try again later.
Mar 06 19:05:53 TWCManager python3[471]: 19:05:53 🚗 TeslaAPI 13 updateCarApiLastErrorTime() called due to Tesla API Error. Updating timestamp from 1709747127.306657 to 1709748353.4
```
When I open the web browser, I see no request to update the "Access/Refresh Token" either. 
In the latest config.json I read about a change on API since 2024 for cars produced after 2021, but my car is a lot older and this all worked before....shouldn't it then work as well now?

Would be nice if someone can help me getting this start/stop command back working.

danielsan1, 2024-03-06:
Hi @GMerg check #563 - maybe the dev branch will help

GMerg, 2024-03-07:
@danielsan1 thanks, this is all very frustrating. I have hopes that @ngardiner or someone else is adding support for the old system again via a fix. Getting an own domain to allow this again is kind of no option.

MikeBishop, 2024-03-07:
If the token is expired, then the refresh token should be used to fetch a new one. An individual token is good for eight hours. Maybe check your logs to see if/why that's failing? It looks something like:

```
Car API auth response{'access_token': 
```

r3pTiLe1860, 2024-03-08:
I am still not lucky:

First i have a **legacy Model S,** so legacy API should still work for me / my car. (I use twcmanager for years now, never hat problems until the versions since the last few months)
Second i am using the **main dev branch** with Update from today (but same behaviour as the weeks before)

I also imported the tokens (Access and Refresh token) via google chrome plugin (Tesla Access Token Generator) and settings page and it correctly writes both tokens to settings.json. 

So now my wrong behaviour. I don‘t get the message as users above (can‘t connect car…) i get the following:

```
Mär 08 16:24:42 raspizero python3[32433]: 16:24:42 ⛽ Manager  20 BackgroundError: Traceback (most recent call last):
Mär 08 16:24:42 raspizero python3[32433]:   File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 271, in background_tasks_thread
Mär 08 16:24:42 raspizero python3[32433]:     carapi.applyChargeLimit(limit=limit, checkArrival=True)
Mär 08 16:24:42 raspizero python3[32433]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 897, in applyChargeLimit
Mär 08 16:24:42 raspizero python3[32433]:     if not vehicle.update_charge() or not vehicle.update_location():
Mär 08 16:24:42 raspizero python3[32433]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1475, in update_charge
Mär 08 16:24:42 raspizero python3[32433]:     return self.update_vehicle_data()
Mär 08 16:24:42 raspizero python3[32433]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1459, in update_vehicle_data
Mär 08 16:24:42 raspizero python3[32433]:     drive = response["drive_state"]
Mär 08 16:24:42 raspizero python3[32433]: KeyError: 'drive_state'
Mär 08 16:24:42 raspizero python3[32433]: , occurred when processing background task
Mär 08 16:24:42 raspizero python3[32433]: 16:24:42 ⛽ Manager  20 Vehicle 5YJSA7H28FFP67223 on TWC 4758 is permitted to charge.
Mär 08 16:24:58 raspizero python3[32433]: 16:24:58 ⛽ Manager  20 Green energy Generates 608W, Consumption 3728W (Charger Load 3931W, Other Load -203W)
```

Unfortunately i have no clue what i can do. I think my tokens are correct, but twcmanager is not able to „use“ them. Please give me a hint what i can check/try?

whisdol, 2024-03-08:
I refreshed the tokens today, will keep an eye on the logs for any access_token errors.
With fresh tokens, I actually receive a 403 error for charging commands, indicating I need to move to the FleetAPI. I haven't been able to dig into the state of support for the FleetAPI in this repo - any pointers on what to do exactly for TWCManager? (I'll start with the Tesla Developer account and [fleetapi-tokens](https://github.com/MikeBishop/fleetapi-tokens))

```log
2024-03-08 12:57:51,815 - 🚗 TeslaAPI 13 Car API cmd vehicles <Response [200]>
2024-03-08 12:57:52,324 - 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/REMOVED <Response [200]>
2024-03-08 12:57:52,350 - 🚗 TeslaAPI 13 CARNAME not ready because it wasn't woken in the last 2 minutes.
2024-03-08 12:57:53,019 - 🚗 TeslaAPI 13 Car API cmd wake_up<Response [200]>
...
2024-03-08 12:58:00,662 - 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
2024-03-08 12:58:01,285 - 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/REMOVED <Response [200]>
2024-03-08 12:58:01,314 - 🚗 TeslaAPI 13 car_api_available returning True
2024-03-08 12:58:01,319 - 🚗 TeslaAPI 13 startOrStop is set to start
2024-03-08 12:58:01,912 - 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/REMOVED/vehicle_data?endpoints=location_data%3Bcharge_state%3Bclimate_state%3Bvehicle_state%3Bgui_settings%3Bvehicle_config <Response [200]>
2024-03-08 12:58:03,161 - 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
2024-03-08 12:58:03,168 - 🚗 TeslaAPI 13 car_api_available returning True
2024-03-08 12:58:06,949 - 🚗 TeslaAPI 13 applyChargeLimit skipped
2024-03-08 12:58:07,477 - 🚗 TeslaAPI 13 Car API cmd charge_start <Response [403]>
2024-03-08 12:58:07,482 - 🚗 TeslaAPI 17 CARNAME: start charge response{'response': None, 'error': 'Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning', 'error_description': ''}
```

r3pTiLe1860, 2024-03-08:
I increased the log level, hope that helps:

```
Mär 08 17:35:09 raspizero python3[1768]: 17:35:09 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Mär 08 17:35:09 raspizero python3[1768]: 17:35:09 🚗 TeslaAPI 13 car_api_available returning True
Mär 08 17:35:10 raspizero python3[1768]: 17:35:10 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/5YJSA7H28FFP67223/vehicle_data?endpoints=location_data%3Bcharge_state%3Bclimate_state%3Bvehicle_state%3Bgui_settings%3Bvehicle_config <Response [200]>
Mär 08 17:35:10 raspizero python3[1768]: 17:35:10 ⛽ Manager  20 BackgroundError: Traceback (most recent call last):
Mär 08 17:35:10 raspizero python3[1768]:   File "/home/pi/TWCManager/lib/TWCManager/TWCManager.py", line 256, in background_tasks_thread
Mär 08 17:35:10 raspizero python3[1768]:     carapi.applyChargeLimit(limit=task["limit"])
Mär 08 17:35:10 raspizero python3[1768]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 897, in applyChargeLimit
Mär 08 17:35:10 raspizero python3[1768]:     if not vehicle.update_charge() or not vehicle.update_location():
Mär 08 17:35:10 raspizero python3[1768]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1475, in update_charge
Mär 08 17:35:10 raspizero python3[1768]:     return self.update_vehicle_data()
Mär 08 17:35:10 raspizero python3[1768]:   File "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py", line 1459, in update_vehicle_data
Mär 08 17:35:10 raspizero python3[1768]:     drive = response["drive_state"]
Mär 08 17:35:10 raspizero python3[1768]: KeyError: 'drive_state'
Mär 08 17:35:10 raspizero python3[1768]: , occurred when processing background task
Mär 08 17:35:10 raspizero python3[1768]: 17:35:10 ⛽ Manager  20 Green energy Generates 120W, Consumption 390W (Charger Load 0W, Other Load 390W)
Mär 08 17:35:10 raspizero python3[1768]: 17:35:10 ⛽ Manager  20 Limiting charging to 0.17A - 0.17A = 0.00A.
Mär 08 17:35:10 raspizero python3[1768]: 17:35:10 ⛽ Manager  20 Charge when above 6A (minAmpsPerTWC).
```

MikeBishop, 2024-03-08:
@whisdol, it looks like the legacy API is now disabled for you, at least as far as commands go. I haven't looked into the Command Protocol much beyond the Fleet API tokens, unfortunately.

@r3pTiLe1860, it looks like the response doesn't have a `drive_state` property. What happens if you run
```
curl --request GET --header 'Authorization: Bearer <<<ACCESS_TOKEN_HERE>>>' https://owner-api.teslamotors.com/api/1/vehicles/<<<VIN_HERE>>>/vehicle_data?endpoints=location_data%3Bcharge_state%3Bclimate_state%3Bvehicle_state%3Bgui_settings%3Bvehicle_config
```

r3pTiLe1860, 2024-03-08:
@MikeBishop thanks for reply.

command executes correctly and gets valid data (but as far as i can see - no drive_state:

`{"response":{"id":<<<ID>>>,"user_id":<<<u_id>>>,"vehicle_id":<<<v_id>>>,"vin":"<<<VIN>>>","color":null,"access_type":"OWNER","granular_access":{"hide_private":false,"training_wheels":false},"tokens":["65611ba3da12cb78","2bccdf72e950cd9f"],"state":"online","in_service":false,"id_s":"<<<ID>>>","calendar_enabled":true,"api_version":36,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false,"charge_state":{"battery_heater_on":false,"battery_level":80,"battery_range":224.52,"charge_amps":6,"charge_current_request":6,"charge_current_request_max":6,"charge_enable_request":false,"charge_energy_added":15.66,"charge_limit_soc":90,"charge_limit_soc_max":100,"charge_limit_soc_min":50,"charge_limit_soc_std":90,"charge_miles_added_ideal":48.5,"charge_miles_added_rated":61.0,"charge_port_cold_weather_mode":null,"charge_port_color":"Off","charge_port_door_open":true,"charge_port_latch":"Engaged","charge_rate":0.0,"charge_to_max_range":false,"charger_actual_current":0,"charger_phases":null,"charger_pilot_current":6,"charger_power":0,"charger_voltage":0,"charging_state":"Stopped","conn_charge_cable":"IEC","est_battery_range":127.84,"fast_charger_brand":"\u003cinvalid\u003e","fast_charger_present":false,"fast_charger_type":"ACSingleWireCAN","ideal_battery_range":179.61,"managed_charging_active":false,"managed_charging_start_time":null,"managed_charging_user_canceled":false,"max_range_charge_counter":0,"minutes_to_full_charge":130,"not_enough_power_to_heat":false,"off_peak_charging_enabled":false,"off_peak_charging_times":"all_week","off_peak_hours_end_time":360,"preconditioning_enabled":false,"preconditioning_times":"all_week","scheduled_charging_mode":"Off","scheduled_charging_pending":false,"scheduled_charging_start_time":null,"scheduled_charging_start_time_app":0,"scheduled_departure_time":null,"supercharger_session_trip_planner":false,"time_to_full_charge":2.17,"timestamp":1709937301135,"trip_charging":false,"usable_battery_level":80,"user_charge_enable_request":false},"climate_state":{"allow_cabin_overheat_protection":false,"battery_heater":false,"battery_heater_no_power":false,"cabin_overheat_protection":"On","climate_keeper_mode":"off","defrost_mode":0,"driver_temp_setting":22.0,"fan_status":0,"hvac_auto_request":"On","inside_temp":13.2,"is_auto_conditioning_on":false,"is_climate_on":false,"is_front_defroster_on":false,"is_preconditioning":false,"is_rear_defroster_on":false,"left_temp_direction":150,"max_avail_temp":28.0,"min_avail_temp":15.0,"outside_temp":7.5,"passenger_temp_setting":22.0,"remote_heater_control_enabled":false,"right_temp_direction":150,"seat_heater_left":0,"seat_heater_right":0,"side_mirror_heaters":false,"supports_fan_only_cabin_overheat_protection":false,"timestamp":1709937301135,"wiper_blade_heater":false},"gui_settings":{"gui_24_hour_time":true,"gui_charge_rate_units":"kW","gui_distance_units":"km/hr","gui_range_display":"Ideal","gui_temperature_units":"C","show_range_units":true,"timestamp":1709937301135},"vehicle_config":{"can_accept_navigation_requests":true,"can_actuate_trunks":true,"car_special_type":"base","car_type":"models","charge_port_type":"EU","dashcam_clip_save_supported":false,"default_charge_to_max":false,"driver_assist":"MonoCam","ece_restrictions":true,"efficiency_package":"Default","eu_vehicle":true,"exterior_color":"SteelGrey","front_drive_unit":"NoneOrSmall","has_air_suspension":true,"has_ludicrous_mode":false,"has_seat_cooling":false,"headlamp_type":"Hid","interior_trim_type":"AllBlack","motorized_charge_port":false,"plg":true,"pws":false,"rear_drive_unit":"Large","rear_seat_heaters":0,"rear_seat_type":0,"rhd":false,"roof_color":"None","seat_type":1,"spoiler_type":"Passive","sun_roof_installed":1,"third_row_seats":"None","timestamp":1709937301136,"trim_badging":"p85d","use_range_badging":false,"utc_offset":3600,"wheel_type":"Slipstream19Carbon"},"vehicle_state":{"api_version":36,"autopark_state_v2":"standby","autopark_style":"dead_man","calendar_supported":true,"car_version":"2022.8.10.17 3563a9c64f1d","center_display_state":0,"dashcam_clip_save_available":false,"dashcam_state":"\u003cinvalid\u003e","df":0,"dr":0,"fd_window":0,"feature_bitmask":"5,0","fp_window":0,"ft":0,"homelink_device_count":1,"homelink_nearby":true,"is_user_present":false,"last_autopark_error":"no_error","locked":true,"media_state":{"remote_control_enabled":false},"notifications_supported":true,"odometer":87603.244639,"parsed_calendar_supported":true,"pf":0,"pr":0,"rd_window":0,"remote_start":false,"remote_start_enabled":true,"remote_start_supported":true,"rp_window":0,"rt":0,"santa_mode":0,"smart_summon_available":false,"software_update":{"download_perc":0,"expected_duration_sec":2700,"install_perc":1,"status":"","version":" "},"speed_limit_mode":{"active":false,"current_limit_mph":85.0,"max_limit_mph":90,"min_limit_mph":50.0,"pin_code_set":false},"summon_standby_mode_enabled":false,"sun_roof_percent_open":0,"sun_roof_state":"closed","timestamp":1709937301135,"tpms_pressure_fl":null,"tpms_pressure_fr":null,"tpms_pressure_rl":null,"tpms_pressure_rr":null,"valet_mode":false,"valet_pin_needed":true,"vehicle_name":"Lion"}}}`

i don't know if there is any critical data, so i replaced some at the beginning. glad to get some help :-)

GMerg, 2024-03-09:
Hello, got some new observations...

> If the token is expired, then the refresh token should be used to fetch a new one. An individual token is good for eight hours. Maybe check your logs to see if/why that's failing? It looks something like:
> 
> ```
> Car API auth response{'access_token': 
> ```

This reply is not showing in my log. After 8 hours the nothing seems to be refreshed. At least all I see in the log is a sudden 401 reply after a 200 one was given shortly before. As log level I have set 13 in config.json.
Below is the log from syslog. I guess you can ignore most of the log entries but maybe helpful after all.
```
Mar  9 07:14:08 TWCManager python3[465]: 07:14:08 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/REMOVED <Response [200]>
Mar  9 07:14:08 TWCManager python3[465]: 07:14:08 🚗 TeslaAPI 13 CAR_NAME not ready because it wasn't woken in the last 2 minutes.
Mar  9 07:14:08 TWCManager python3[465]: 07:14:08 ⛽ Manager  20 Green energy Generates 0W, Consumption 178W (Charger Load 0W, Other Load 178W)
Mar  9 07:14:08 TWCManager python3[465]: 07:14:08 ⛽ Manager  20 Limiting charging to 0.00A - 0.00A = 0.00A.
Mar  9 07:14:08 TWCManager python3[465]: 07:14:08 ⛽ Manager  20 Charge when above 6A (minAmpsPerTWC).
Mar  9 07:14:22 TWCManager python3[465]: 07:14:22 ⛽ Manager  20 Slave TWC 0441: Delivered 11554 kWh, voltage per phase: (238, 0, 0).
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (settings.chargeNowAmps [0]), condition (gt), value (0)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (checkScheduledCharging() [0]), condition (eq), value (1)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (tm_hour [7]), condition (gte), value (7)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (tm_hour [7]), condition (lt), value (18)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (settings.hourResumeTrackGreenEnergy [0]), condition (lte), value (7)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛅ TeslaPwl 15 Powerwall2 API token still valid for 2045.729723930359 seconds.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (modules.TeslaPowerwall2.generatedW [184.0]), condition (gte), value (1500)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛅ TeslaPwl 15 Powerwall2 API token still valid for 2045.6488015651703 seconds.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (modules.TeslaPowerwall2.batteryLevel [-0.03968628933194782]), condition (gte), value (45)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (modules.TeslaPowerwall2.exportW [15.0]), condition (gt), value (200)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (modules.TeslaAPI.minBatteryLevelAtHome [90]), condition (lt), value (40)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (modules.TeslaAPI.minBatteryLevelAtHome [90]), condition (lt), value (45)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (modules.TeslaAPI.minBatteryLevelAtHome [90]), condition (lt), value (80)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (modules.TeslaAPI.minBatteryLevelAtHome [90]), condition (lt), value (80)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (modules.TeslaAPI.minBatteryLevelAtHome [90]), condition (lt), value (80)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (settings.nonScheduledAction [3]), condition (lt), value (3)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Policy conditions were not matched.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   13 Evaluating Policy match (settings.nonScheduledAction [3]), condition (eq), value (3)
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛽ Policy   14 All policy conditions have matched. Policy chosen is Track Green Energy
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 ⛅ TeslaPwl 16 Powerwall needs to charge. Ignoring generation.
Mar  9 07:14:38 TWCManager python3[465]: 07:14:38 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
[... cut ...]
Mar  9 07:15:09 TWCManager python3[465]: 07:15:09 🚗 TeslaAPI 13 car_api_available returning True
Mar  9 07:15:09 TWCManager python3[465]: 07:15:09 🚗 TeslaAPI 13 Car API cmd https://owner-api.teslamotors.com/api/1/vehicles/REMOVED <Response [401]>
Mar  9 07:15:09 TWCManager python3[465]: 07:15:09 🚗 TeslaAPI 20 ERROR: Can't access vehicle status for CAR_NAME.  Will try again later.
Mar  9 07:15:09 TWCManager python3[465]: 07:15:09 🚗 TeslaAPI 13 updateCarApiLastErrorTime() called due to Tesla API Error. Updating timestamp from 0 to 1709964909.4533992

```

Flitzer42, 2024-03-09:
@GMerg I am still facing the same issues. Exactly the same log as yours. Car is M3 2019. Also no request to re-enter the tokens. But when restarting, the request appears.
When the token expires, nothing happens, but when looking in the "settings.json", the refresh token is empty.
I assume, with the empty refresh token, it will not work.
@MikeBishop here is a [project ](https://github.com/snaptec/openWB/tree/master/modules/soc_tesla) which uses the TeslaAPI to request the SoC. They had the same issues in December, but where able to fix it. The procedure to get a token is similar. Perhaps this can provide some inspiration.

MikeBishop, 2024-03-10:
@Flitzer42, it's not that we don't have code to refresh the token. If it's failing for some reason, we'll need to understand why.

@GMerg, going out on a limb, but what if you instead do:
```
curl --request GET --header 'Authorization: Bearer <<<ACCESS_TOKEN_HERE>>>' https://owner-api.teslamotors.com/api/1/vehicles/<<<VIN_HERE>>>/vehicle_data?endpoints=location_data%3Bdrive_state%3Bcharge_state%3Bclimate_state%3Bvehicle_state%3Bgui_settings%3Bvehicle_config
```

Previously `location_data` implied `drive_state`, but I'm wondering if old cars need to ask for it explicitly now?  (We could probably trim that list a bit, in fact, but that's a separate issue.)

r3pTiLe1860, 2024-03-10:
@MikeBishop i think you meant me regarding drive_state.

I think we get further - looks better at least for me:

`{"response":{"id":<<<ID>>>,"user_id":<<<u_id>>>,"vehicle_id":<<<v_id>>>,"vin":"<<<VIN>>>","color":null,"access_type":"OWNER","granular_access":{"hide_private":false,"training_wheels":false},"tokens":["e15547fdeac11a58","ce0330c59b8c0815"],"state":"online","in_service":false,"id_s":"<<<ID>>>","calendar_enabled":true,"api_version":36,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false,"charge_state":{"battery_heater_on":false,"battery_level":72,"battery_range":198.61,"charge_amps":32,"charge_current_request":32,"charge_current_request_max":32,"charge_enable_request":true,"charge_energy_added":47.01,"charge_limit_soc":85,"charge_limit_soc_max":100,"charge_limit_soc_min":50,"charge_limit_soc_std":90,"charge_miles_added_ideal":146.0,"charge_miles_added_rated":182.5,"charge_port_cold_weather_mode":null,"charge_port_color":"Off","charge_port_door_open":false,"charge_port_latch":"Blocking","charge_rate":0.0,"charge_to_max_range":false,"charger_actual_current":0,"charger_phases":null,"charger_pilot_current":32,"charger_power":0,"charger_voltage":0,"charging_state":"Disconnected","conn_charge_cable":"\u003cinvalid\u003e","est_battery_range":158.51,"fast_charger_brand":"\u003cinvalid\u003e","fast_charger_present":false,"fast_charger_type":"\u003cinvalid\u003e","ideal_battery_range":158.89,"managed_charging_active":false,"managed_charging_start_time":null,"managed_charging_user_canceled":false,"max_range_charge_counter":0,"minutes_to_full_charge":0,"not_enough_power_to_heat":false,"off_peak_charging_enabled":false,"off_peak_charging_times":"all_week","off_peak_hours_end_time":360,"preconditioning_enabled":false,"preconditioning_times":"all_week","scheduled_charging_mode":"Off","scheduled_charging_pending":false,"scheduled_charging_start_time":null,"scheduled_charging_start_time_app":0,"scheduled_departure_time":null,"supercharger_session_trip_planner":false,"time_to_full_charge":0.0,"timestamp":1710095584166,"trip_charging":false,"usable_battery_level":71,"user_charge_enable_request":null},"climate_state":{"allow_cabin_overheat_protection":false,"battery_heater":false,"battery_heater_no_power":false,"cabin_overheat_protection":"On","climate_keeper_mode":"off","defrost_mode":0,"driver_temp_setting":22.0,"fan_status":0,"hvac_auto_request":"On","inside_temp":null,"is_auto_conditioning_on":null,"is_climate_on":false,"is_front_defroster_on":false,"is_preconditioning":false,"is_rear_defroster_on":false,"left_temp_direction":null,"max_avail_temp":28.0,"min_avail_temp":15.0,"outside_temp":null,"passenger_temp_setting":22.0,"remote_heater_control_enabled":false,"right_temp_direction":null,"seat_heater_left":0,"seat_heater_right":0,"side_mirror_heaters":false,"supports_fan_only_cabin_overheat_protection":false,"timestamp":1710095584165,"wiper_blade_heater":false},"drive_state":{"gps_as_of":1710095582,"heading":233,"latitude":49.448826,"longitude":11.815098,"native_latitude":49.448826,"native_location_supported":1,"native_longitude":11.815098,"native_type":"wgs","power":0,"shift_state":null,"speed":null,"timestamp":1710095584166},"gui_settings":{"gui_24_hour_time":true,"gui_charge_rate_units":"kW","gui_distance_units":"km/hr","gui_range_display":"Ideal","gui_temperature_units":"C","show_range_units":true,"timestamp":1710095584166},"vehicle_config":{"can_accept_navigation_requests":true,"can_actuate_trunks":true,"car_special_type":"base","car_type":"models","charge_port_type":"EU","dashcam_clip_save_supported":false,"default_charge_to_max":false,"driver_assist":"MonoCam","ece_restrictions":true,"efficiency_package":"Default","eu_vehicle":true,"exterior_color":"SteelGrey","front_drive_unit":"NoneOrSmall","has_air_suspension":true,"has_ludicrous_mode":false,"has_seat_cooling":false,"headlamp_type":"Hid","interior_trim_type":"AllBlack","motorized_charge_port":false,"plg":true,"pws":false,"rear_drive_unit":"Large","rear_seat_heaters":0,"rear_seat_type":0,"rhd":false,"roof_color":"None","seat_type":1,"spoiler_type":"Passive","sun_roof_installed":1,"third_row_seats":"None","timestamp":1710095584166,"trim_badging":"p85d","use_range_badging":false,"utc_offset":3600,"wheel_type":"Slipstream19Carbon"},"vehicle_state":{"api_version":36,"autopark_state_v2":"ready","autopark_style":"dead_man","calendar_supported":true,"car_version":"2022.8.10.17 3563a9c64f1d","center_display_state":0,"dashcam_clip_save_available":false,"dashcam_state":"\u003cinvalid\u003e","df":0,"dr":0,"fd_window":0,"feature_bitmask":"5,0","fp_window":0,"ft":0,"homelink_device_count":1,"homelink_nearby":true,"is_user_present":false,"last_autopark_error":"no_error","locked":true,"media_state":{"remote_control_enabled":false},"notifications_supported":true,"odometer":87833.203587,"parsed_calendar_supported":true,"pf":0,"pr":0,"rd_window":0,"remote_start":false,"remote_start_enabled":true,"remote_start_supported":true,"rp_window":0,"rt":0,"santa_mode":0,"smart_summon_available":false,"software_update":{"download_perc":0,"expected_duration_sec":2700,"install_perc":1,"status":"","version":" "},"speed_limit_mode":{"active":false,"current_limit_mph":85.0,"max_limit_mph":90,"min_limit_mph":50.0,"pin_code_set":false},"summon_standby_mode_enabled":false,"sun_roof_percent_open":0,"sun_roof_state":"closed","timestamp":1710095584165,"tpms_pressure_fl":null,"tpms_pressure_fr":null,"tpms_pressure_rl":null,"tpms_pressure_rr":null,"valet_mode":false,"valet_pin_needed":true,"vehicle_name":"Lion"}}}`

GMerg, 2024-03-10:
> curl --request GET --header 'Authorization: Bearer <<<ACCESS_TOKEN_HERE>>>' https://owner-api.teslamotors.com/api/1/vehicles/<<<VIN_HERE>>>/vehicle_data?endpoints=location_data%3Bdrive_state%3Bcharge_state%3Bclimate_state%3Bvehicle_state%3Bgui_settings%3Bvehicle_config

I get the following:
`{"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}`

r3pTiLe1860, 2024-03-10:
@GMerg you have to wake up the car before. Just connect with app and repeat the command

MikeBishop, 2024-03-11:
> @MikeBishop i think you meant me regarding drive_state :-).
> 
> I think we get further - looks better at least for me:

Great! I made #568 to add that scope, since I don't think there's any harm in doing it generally.

r3pTiLe1860, 2024-03-11:
@MikeBishop thank you so much, looking forward to the merge and i will report then immediately :-)

GMerg, 2024-03-11:
@r3pTiLe1860 I have now the log after waking it up and performing the command @MikeBishop wanted...
`response":"id":1xxxxxxxxxxxxxxx,"user_id":3xxxxx,"vehicle_id":2xxxxxxxxx,"vin":">REMOVED>","color":null,"access_type":"OWNER","granular_access":{"hide_private":false,"training_wheels":false},"tokens":["<8bytehexstring>","<8bytehexstring>"],"state":"online","in_service":false,"id_s":"1xxxxxxxxxxxxxxx","calendar_enabled":true,"api_version":73,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false,"charge_state":{"battery_heater_on":false,"battery_level":90,"battery_range":310.24,"charge_amps":6,"charge_current_request":6,"charge_current_request_max":6,"charge_enable_request":true,"charge_energy_added":4.22,"charge_limit_soc":90,"charge_limit_soc_max":100,"charge_limit_soc_min":50,"charge_limit_soc_std":90,"charge_miles_added_ideal":12.0,"charge_miles_added_rated":15.0,"charge_port_cold_weather_mode":null,"charge_port_color":"Green","charge_port_door_open":true,"charge_port_latch":"Engaged","charge_rate":0.0,"charger_actual_current":0,"charger_phases":null,"charger_pilot_current":6,"charger_power":0,"charger_voltage":0,"charging_state":"Complete","conn_charge_cable":"IEC","est_battery_range":234.07,"fast_charger_brand":"\u003cinvalid\u003e","fast_charger_present":false,"fast_charger_type":"ACSingleWireCAN","ideal_battery_range":248.48,"max_range_charge_counter":0,"minutes_to_full_charge":0,"not_enough_power_to_heat":false,"off_peak_charging_enabled":false,"off_peak_charging_times":"all_week","off_peak_hours_end_time":360,"preconditioning_enabled":false,"preconditioning_times":"all_week","scheduled_charging_mode":"Off","scheduled_charging_pending":false,"scheduled_charging_start_time":null,"scheduled_charging_start_time_app":0,"scheduled_departure_time":1704900600,"scheduled_departure_time_minutes":990,"supercharger_session_trip_planner":false,"time_to_full_charge":0.0,"timestamp":1710193379302,"trip_charging":false,"usable_battery_level":90,"user_charge_enable_request":null},"climate_state":{"allow_cabin_overheat_protection":false,"battery_heater":false,"battery_heater_no_power":false,"bioweapon_mode":false,"cabin_overheat_protection":"Off","climate_keeper_mode":"off","cop_activation_temperature":"High","defrost_mode":0,"driver_temp_setting":22.0,"fan_status":0,"hvac_auto_request":"Override","inside_temp":8.6,"is_auto_conditioning_on":false,"is_climate_on":false,"is_front_defroster_on":false,"is_preconditioning":false,"is_rear_defroster_on":false,"left_temp_direction":177,"max_avail_temp":28.0,"min_avail_temp":15.0,"outside_temp":6.0,"passenger_temp_setting":21.0,"remote_heater_control_enabled":false,"right_temp_direction":148,"seat_heater_left":0,"seat_heater_rear_center":0,"seat_heater_rear_left":0,"seat_heater_rear_right":0,"seat_heater_right":0,"seat_heater_third_row_left":0,"seat_heater_third_row_right":0,"side_mirror_heaters":false,"steering_wheel_heater":false,"supports_fan_only_cabin_overheat_protection":false,"timestamp":1710193379301,"wiper_blade_heater":false},"drive_state":{"active_route_latitude":xx.xxx696,"active_route_longitude":xx.xxx3742,"active_route_traffic_minutes_delay":0.0,"gps_as_of":1710193377,"heading":307,"latitude":xx.xxx686,"longitude":xx.xxx704,"native_latitude":xx.xxx686,"native_location_supported":1,"native_longitude":xx.xxx704,"native_type":"wgs","power":0,"shift_state":"P","speed":null,"timestamp":1710193379302},"gui_settings":{"gui_24_hour_time":true,"gui_charge_rate_units":"km/hr","gui_distance_units":"km/hr","gui_range_display":"Ideal","gui_temperature_units":"C","gui_tirepressure_units":"Bar","show_range_units":true,"timestamp":1710193379302},"vehicle_config":{"can_accept_navigation_requests":true,"can_actuate_trunks":true,"car_special_type":"base","car_type":"modelx","charge_port_type":"EU","cop_user_set_temp_supported":false,"dashcam_clip_save_supported":true,"default_charge_to_max":false,"driver_assist":"TeslaAP3","ece_restrictions":true,"efficiency_package":"Default","eu_vehicle":true,"exterior_color":"MetallicBlack","exterior_trim_override":"Chrome","front_drive_unit":"NoneOrSmall","has_air_suspension":true,"has_ludicrous_mode":false,"has_seat_cooling":false,"headlamp_type":"Led","interior_trim_type":"AllBlack","motorized_charge_port":true,"paint_color_override":"14,14,14,1,0.04","plg":true,"pws":false,"rear_drive_unit":"Small","rear_seat_heaters":3,"rear_seat_type":7,"rhd":false,"roof_color":"None","seat_type":0,"spoiler_type":"Passive","sun_roof_installed":0,"supports_qr_pairing":false,"third_row_seats":"FuturisFoldFlat","timestamp":1710193379302,"trim_badging":"100d","use_range_badging":false,"utc_offset":3600,"webcam_selfie_supported":false,"webcam_supported":false,"wheel_type":"AeroTurbine20"},"vehicle_state":{"allow_authorized_mobile_devices_only":false,"api_version":73,"autopark_state_v2":"disabled","autopark_style":"dead_man","calendar_supported":true,"car_version":"2024.2.7d522c44937f7","center_display_state":0,"dashcam_clip_save_available":true,"dashcam_state":"Recording","df":0,"dr":0,"fd_window":0,"feature_bitmask":"f9dffbff,2987f","fp_window":0,"ft":0,"homelink_device_count":0,"homelink_nearby":false,"is_user_present":false,"last_autopark_error":"no_error","locked":true,"media_info":{"audio_volume":2.6667,"audio_volume_increment":0.333333,"audio_volume_max":10.333333,"media_playback_status":"Stopped","now_playing_album":"","now_playing_artist":"","now_playing_duration":0,"now_playing_elapsed":0,"now_playing_source":"T [... cut ...] pms_last_seen_pressure_time_rl":1709995436,"tpms_last_seen_pressure_time_rr":1709995436,"tpms_pressure_fl":2.85,"tpms_pressure_fr":2.85,"tpms_pressure_rl":2.85,"tpms_pressure_rr":2.825,"tpms_rcp_front_value":2.9,"tpms_rcp_rear_value":2.9,"tpms_soft_warning_fl":false,"tpms_soft_warning_fr":false,"tpms_soft_warning_rl":false,"tpms_soft_warning_rr":false,"valet_mode":false,"valet_pin_needed":true,"vehicle_name":"XYZ","webcam_available":false}}}
`
I hope I have removed every sensitive data there is.

r3pTiLe1860, 2024-03-12:
@MikeBishop thanks again for the fix. I checked after merge on main and it worked :-) logging looks good, plugged in car at night and the 6amp charging stopped after few moments. 

Now i hope the handling of access/refresh token works correctly. Will check again tomorrow.

GMerg, 2024-03-13:
The access/refresh still seems to be a problem, at least for me.

r3pTiLe1860, 2024-03-13:
I can confirm that. Behavior is the same for me: Yesterday i updated and restarted - had to put in new tokens via URL method and all worked well. But as @GMerg stated there seems to be no update. After the token expires no more connection over API is possible and the refresh token is removed from settings.json.

Doesn‘t seem to belong to fleet api because for my case it‘s the owner API.

GMerg, 2024-03-13:
Thanks for confirming! I got same issues as @Flitzer42 and @r3pTiLe1860 reported.

Heptagon404, 2024-03-29:
Same problem here M3 2020 and MY 2022. If I manually fill in the tokens it works for ~ a day. Then the api control gets lost. Im on v1.3.2.

r3pTiLe1860, 2024-03-29:
@MikeBishop or @ngardiner please help 🙏

dtiefnig, 2024-04-01:
Same for me since I upgraded to 1.3.2.

Before the upgrade I can see the token refresh in the logs every 7 hours. It returns with `Car API auth response ...` and lists `access_token`, `refresh_token` and `id_token`.

After the upgrade to 1.3.2, the next refresh only lists `access_token` and `id_token`. No `refresh_token` at all.

In the `apiRefresh()` function then `self.setCarApiRefreshToken()` fails because the token is missing.

After that, no refresh is tried anymore (probably we should change the code to retry the refresh with the old token) and when I try to access the API now it says `{"error":"token expired (401)"}` of course. Also `carApiRefreshToken` in `/etc/twcmanager/settings.json` is empty, so the old refresh token got erased.

Trying to reproduce that manually to see what is going on. If you have any ideas it would be much appreciated. Did not spend a lot of time with the API so far.

dtiefnig, 2024-04-01:
Okay, fortunately refresh tokens are valid for quite some time. When I'm calling

> `curl --request POST https://auth.tesla.com/oauth2/v3/token --json '{"grant_type": "refresh_token", "client_id": "ownerapi", "refresh_token": "<<<LAST-RETURNED-REFRESH-TOKEN>>>"}'`

I'm indeed getting a response without a refresh token:

> `{"access_token":"<<<NEW-ACCESS-TOKEN>>>","id_token":"<<<NEW-ID-TOKEN>>>","expires_in":28800,"token_type":"Bearer"}`

Is this expected behavior currently? twcmanager at least seems to expect a new refresh token in the response.

Did they disable token refresh to force people to the fleet API?

r3pTiLe1860, 2024-04-01:
Thanks @dtiefnig for investigating. Would be absolutely great if that fixes the token refresh. Fingers also crossed that owner api will stay for legacy cars ;-)

dtiefnig, 2024-04-01:
Works for me currently with my 2019 M3P. Hope it stays that way for the time being.

Flitzer42, 2024-04-02:
hi @dtiefnig , thanks for the investigation. I added the <"scope": "offline_access"> as stated in https://github.com/ngardiner/TWCManager/pull/572, but still, the <"carApiRefreshToken": ""> is empty in the settings.json after attempting to refresh the token. What else did you change? I am on a 2019 M3LR.

dtiefnig, 2024-04-02:
@Flitzer42 can you run the curl command with a valid refresh token? Does it return a new refresh token?

> `curl --request POST https://auth.tesla.com/oauth2/v3/token --json '{"scope": "offline_access", "grant_type": "refresh_token", "client_id": "ownerapi", "refresh_token": "<<<LAST-RETURNED-REFRESH-TOKEN>>>"}'`

Did not have to change anything else to get this working again.

Flitzer42, 2024-04-02:
@dtiefnig thanks for for response and support. I am running the TWCManager on a raspberry with buster (I failed to get TWCManager running on a bullseye. ). The curl command gives me a "curl: option --json: is unknown" as response. So something else is missing. I am no Linux or programming expert...
BTW: When starting TWCMAnager and entering the URL from Teslas authentication page, the settings.json contains both, the bearer and the refresh token (it works fine until the refresh). But as soon as TWCMAnager attempts to refresh, the refresh token disappears from the file.

dtiefnig, 2024-04-02:
Ah, the --json is just a convenience option, try the following:

> `curl --request POST https://auth.tesla.com/oauth2/v3/token --header 'Content-Type: application/json' -d '{"scope": "offline_access", "grant_type": "refresh_token", "client_id": "ownerapi", "refresh_token": "<<<YOUR-REFRESH-TOKEN"}'`

Flitzer42, 2024-04-02:
Thanks for clarification, The above request gives me a refresh token. That's fine. I did not test the function of the application again (takes some time until it refreshes the token).  Let's see tomorrow....

Flitzer42, 2024-04-03:
Logged the refresh attempt with journalctl:
`Apr 03 14:53:40 raspberrypi python3[437]: 14:53:40 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
Apr 03 14:53:40 raspberrypi python3[437]: 14:53:40 🚗 TeslaAPI 13 Attempting token refresh
Apr 03 14:53:40 raspberrypi python3[437]: 14:53:40 🚗 TeslaAPI 19 Car API request<Response [200]>
Apr 03 14:53:40 raspberrypi python3[437]: 14:53:40 🚗 TeslaAPI 17 Car API auth response{'access_token': '<<Token>>', 'id_token': '<Token>', 'expires_in': 28800, 'token_type': 'Bearer'}
Apr 03 14:53:40 raspberrypi python3[437]: --- Logging error ---
Apr 03 14:53:40 raspberrypi python3[437]: Traceback (most recent call last):
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager/Vehicle/TeslaAPI.py", line 150, in apiRefresh
Apr 03 14:53:40 raspberrypi python3[437]:     self.setCarApiRefreshToken(apiResponseDict["refresh_token"])
Apr 03 14:53:40 raspberrypi python3[437]: KeyError: 'refresh_token'
Apr 03 14:53:40 raspberrypi python3[437]: During handling of the above exception, another exception occurred:
Apr 03 14:53:40 raspberrypi python3[437]: Traceback (most recent call last):
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/lib/python3.7/logging/__init__.py", line 1034, in emit
Apr 03 14:53:40 raspberrypi python3[437]:     msg = self.format(record)
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/lib/python3.7/logging/__init__.py", line 880, in format
Apr 03 14:53:40 raspberrypi python3[437]:     return fmt.format(record)
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager/Logging/ConsoleLogging.py", line 19, in format
Apr 03 14:53:40 raspberrypi python3[437]:     s = super(ColorFormatter, self).format(record)
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/lib/python3.7/logging/__init__.py", line 619, in format
Apr 03 14:53:40 raspberrypi python3[437]:     record.message = record.getMessage()
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/lib/python3.7/logging/__init__.py", line 380, in getMessage
Apr 03 14:53:40 raspberrypi python3[437]:     msg = msg % self.args
Apr 03 14:53:40 raspberrypi python3[437]: TypeError: not all arguments converted during string formatting
Apr 03 14:53:40 raspberrypi python3[437]: Call stack:
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/lib/python3.7/threading.py", line 885, in _bootstrap
Apr 03 14:53:40 raspberrypi python3[437]:     self._bootstrap_inner()
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Apr 03 14:53:40 raspberrypi python3[437]:     self.run()
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/lib/python3.7/threading.py", line 865, in run
Apr 03 14:53:40 raspberrypi python3[437]:     self._target(*self._args, **self._kwargs)
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager/TWCManager.py", line 256, in background_tasks_thread
Apr 03 14:53:40 raspberrypi python3[437]:     carapi.applyChargeLimit(limit=task["limit"])
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager/Vehicle/TeslaAPI.py", line 817, in applyChargeLimit
Apr 03 14:53:40 raspberrypi python3[437]:     if not self.car_api_available():
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager/Vehicle/TeslaAPI.py", line 221, in car_api_available
Apr 03 14:53:40 raspberrypi python3[437]:     self.apiRefresh()
Apr 03 14:53:40 raspberrypi python3[437]:   File "/usr/local/lib/python3.7/dist-packages/TWCManager/Vehicle/TeslaAPI.py", line 158, in apiRefresh
Apr 03 14:53:40 raspberrypi python3[437]:     "ERROR: Can't access Tesla car via API.  Please log in again via web interface.",
Apr 03 14:53:40 raspberrypi python3[437]: Message: 'TeslaAPI'
Apr 03 14:53:40 raspberrypi python3[437]: Arguments: ("ERROR: Can't access Tesla car via API.  Please log in again via web interface.",)
`
Any Idea?
Think the API response is "ID Token" and the missing refresh token leads to the further issues.
Manual curl still works...

dtiefnig, 2024-04-03:
Hmm, according to the log, the refresh token is not returned. Are you sure the code was patched correctly?

Did you restart twcmanager to reload the source code? This does not happen automatically.

If you restarted, is the patch still applied after the restart? Might be something is reverting the code back to the old version.

Flitzer42, 2024-04-03:
Sure, rasperry restartet and verified with WinSCP and nano that the patch is in. (Was my first thought too :-) )
`    def apiRefresh(self):
        # Refresh tokens expire in 45
        # days when first issued, so we'll get a new token every 15 days.
        headers = {"accept": "application/json", "Content-Type": "application/json"}
        data = {
            "client_id": self.refreshClientID,
            "grant_type": "refresh_token",
            "refresh_token": self.getCarApiRefreshToken(),
            "scope": "offline_access",
        }
        req = None
        now = time.time()
`
The comma at the end of the last line makes me curious...
Is there a way to see the request that i sent to the API to verify?

dtiefnig, 2024-04-03:
You can try the following patch:
```
--- a/lib/TWCManager/Vehicle/TeslaAPI.py
+++ b/lib/TWCManager/Vehicle/TeslaAPI.py
@@ -130,7 +130,8 @@ class TeslaAPI:
         now = time.time()
         try:
             req = requests.post(self.refreshURL, headers=headers, json=data)
-            logger.log(logging.INFO2, "Car API request" + str(req))
+            logger.log(logging.INFO2, "Car API request " + self.refreshURL + ", headers = " + str(headers) + ", json = " + str(data))
+            logger.log(logging.INFO2, "Car API response " + str(req))
             apiResponseDict = json.loads(req.text)
         except requests.exceptions.RequestException:
             logger.log(
```
Apply with `patch -p1` or use your favorite editor to replace the `logger.log` line which logs the response with the two `logger.log` lines that log request and response.

Maybe a good opportunity to fix the typo in mainline as well, as the log line currently states `Car API request` but logs the response, which is a little confusing...

Flitzer42, 2024-04-05:
Thanks for the support. I will do some further investigations on the weekend. Seems that the section starting with "try: ..." is not executed. Need to add further log outputs to trace. Will come back to you later.

dtiefnig, 2024-04-05:
Oh, would make sense to put the new logger log line for the request *before* the `requests.post` line actually. It should then in any case log the request even if the post() bails out ...

Flitzer42, 2024-04-05:
Yea, that's the plan
Is there a trick to speed up refresh from 7 hours to ~ 15 minutes for testing purpose?

dtiefnig, 2024-04-05:
Sure, it's just software. :)

The code is a little further down in the file. It checks whether the expiration timestamp of the token is less than x seconds in the future. As the tokens are valid for 8 hours, the 60 * 465 will cause the token to refresh 15 minutes after they have been issued. I'm using that myself to test the renewal.

```
--- a/lib/TWCManager/Vehicle/TeslaAPI.py
+++ b/lib/TWCManager/Vehicle/TeslaAPI.py
@@ -204,7 +206,7 @@ class TeslaAPI:
         # Authentiate to Tesla API
         if not self.master.tokenSyncEnabled() and (
             self.getCarApiBearerToken() == ""
-            or self.getCarApiTokenExpireTime() - now < 60 * 60
+            or self.getCarApiTokenExpireTime() - now < 60 * 465
         ):
             if self.getCarApiRefreshToken() != "":
                 headers = {
```

Flitzer42, 2024-04-05:
Ohh shit, was working all the time on: "/home/pi/TWCManager/lib/TWCManager/Vehicle/TeslaAPI.py". This had no effect.
When adapting: "/usr/local/lib/python3.7/dist-packages/TWCManager/Vehicle/TeslaAPI.py" it works fine.
--> Your solution is approved and I need to learn why I have two instance of the file on my system
Thanks for your great support. Hope the adaptations make it into the release.

dtiefnig, 2024-04-05:
Hah, a classic. So I think this issue could be closed.

The original problem was fixed anyway by @MikeBishop already, so if there are issues with the token handling still, it may be better to open a new issue. Works for you also @GMerg?

r3pTiLe1860, 2024-04-05:
I can also confirm the fix. Many thanks to all of you.

I updated to main after the refresh token and the let the car sleep fix were merged. Now, two days later the car was started to charge, the token is refeshed, all works perfectly :-)

GMerg, 2024-04-05:
@Flitzer42 seems as I made the same mistake and run a different/old version after update. Now Im running the real deal and will have to test for 8 hours till the refresh token is about to get renewed. I will report back if that worked.

GMerg, 2024-04-07:
I can confirm that the latest update fixes things with three notes that you need to follow to get it working:
1. Make sure you use the right installation (use **sudo git pull** followed by **make install**), Without the **make install** you are running the old version
2. Do NOT enter the access/refresh token manually from the setting page. That will still get it working for just 8hours before failing again.
3. You need to get the access/state variable content from the tesla.com page using the login link from the web interface on the main page of your TWCManager. BUT before pasting the link, you need to modify it till #566 is fixed as well.

r3pTiLe1860, 2024-04-07:
Note 3: Interesting observation. For me it works without customization (legacy car P85D) I was able to paste the link from Tesla and it (refresh) works. Maybe a difference for legacy cars?

GMerg, 2024-04-08:
Two days now and all working perfectly with my additions above. So Maybe a fix on #566 and maybe removing the manual access/refresh data (or add a note that state is needed as well) will fix it for everyone.

closed 2024-04-08 by GMerg