ngardiner/TWCManager #566: regular expression in line 1050 of Vehicle/TeslaAPI.py needs to be changed

GMerg, 2024-03-07:
I have noticed a bug which might be helpful in this case:

original code (line 1050 in Vehicle/TeslaAPI.py)
```
        if isinstance(url, bytes):
            url = url.decode("UTF-8")
        code = re.search(r"code=(.+)&state=(.+)", url)

```
This assumes that after the "state" variable, nothing is there. But that was not the case for me. It had additinal parameters in it that got added as "state" later on and so the requests always failed. This regular expression has to optimized to skip additinal parameters. I also would suggest to change it in case the reply changes (state before code in URL).

_Originally posted by @GMerg in https://github.com/ngardiner/TWCManager/issues/564#issuecomment-1983178338_

whisdol, 2024-03-07:
This certainly explains why pasting the URL never works and the manual access & refresh token input is required.

My regex prowess is limited, but [Line 1053](https://github.com/ngardiner/TWCManager/blob/main/lib/TWCManager/Vehicle/TeslaAPI.py#L1053C9-L1053C55) should then be in the simplest case:
```python
code2 = re.search(r"code=(.+)&state=(.+)&*", url)
```

But it's a better idea to make this more generic and just read `code` and `state`  separately.

dtiefnig, 2024-04-13:
There are sure better ways to write this, but as the value of the state parameter (i.e. `code.group(2)` as returned by `re` and written to the log) is never used, what difference does it make? Is there really an issue with saving the token? Might be something else then.