ngardiner/TWCManager #576: A "disconnected" car is never charged when it is plugged in later

RichieB2B, 2024-04-25:
Now that #573 is fixed the logic flaw is back. Yesterday the sun was shining and the "Track Green Energy" policy was active. My wife came home and plugging the charger. She called me 10 minutes later to ask why the car was not charging. The logs show the reason: TWCManager was too fast and sent the charge_start cmd before the car was plugged in. It sets `stopAskingToStartCharging = True` and never retries.

https://github.com/ngardiner/TWCManager/blob/2a6a914657e5414ef96987b1184333c77bc22e75/lib/TWCManager/Vehicle/TeslaAPI.py#L726-L732

"disconnected" is treated the same as "complete", "charging", "is_charging" and "requested" which IMHO is wrong. A car may be "disconnected" now but it can be plugged in later. I'll see if I can fix this flaw.

```
2024-04-24 10:29:39,531 - 🚗 TeslaAPI 20 Big Red is not at home.  Do not start charge.
2024-04-24 10:48:25,204 - 🚗 TeslaAPI 19 Big Red has arrived
2024-04-24 10:48:30,528 - 🚗 TeslaAPI 13 Car API cmd charge_start <Response [200]>
2024-04-24 10:48:30,529 - 🚗 TeslaAPI 17 Big Red: start charge response{'response': {'reason': 'disconnected', 'result': False}}
2024-04-24 10:48:30,530 - 🚗 TeslaAPI 20 Big Red is done charging or already trying to charge or not connected to a charger.  Stop asking to start charging.
2024-04-24 10:49:30,467 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:50:30,282 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:51:30,091 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:52:30,029 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:53:30,860 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:54:30,688 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:55:30,466 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:56:30,344 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:57:30,026 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:58:30,591 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
```

MikeBishop, 2024-04-29:
I'm the one who added "disconnected" to this case, IIRC. The reasoning was that when the EVSE is offering power and the car is plugged in and below its charge limit, it will start drawing power. There's no need to tell the car to start charging in that case. I'm actually more curious about the case where your wife plugged the car in and it didn't start charging spontaneously.

However, if you don't clear it, it keeps the car awake by constantly telling it to start charging when it can't.  That's also my concern with the approach in #577, where the disconnected state is checked prior to sending the command. Does this permit the car to sleep when it's not plugged in, or is it constantly pinging the car to check whether it's connected yet? (Note that this concern would be moot if using Teslamate telemetry, since TWCManager won't be directly pinging the car in that case.)

RichieB2B, 2024-04-29:
You make some good points. In my test with #577 applied my car Kriek started charging all by itself. TWCManager did not have to start it:

```
2024-04-26 08:44:40,279 - 🚗 TeslaAPI 20 Kriek is disconnected.  Do not start charge.
...
2024-04-26 08:45:05,727 - ⛽ Master   20 Charge Session Started for Slave TWC 2119
```

Why plugging in Big Red in [this log](https://github.com/ngardiner/TWCManager/issues/576#issue-2264365087) did not start a charge session is the big mystery. She managed to get the charge started by power cycling the TWC.

```
2024-04-24 10:57:53,587 - ⛽ Slave    20 SHB 2119: 00 00.00/00.00A 0000 0000  M: 09 00.00/24.00A 0000 0000
2024-04-24 10:58:25,738 - ⛽ Slave    20 SHB 2119: 09 00.00/24.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
2024-04-24 10:58:30,589 - 🚗 TeslaAPI 13 Entering car_api_available - next step is to query Tesla API
2024-04-24 10:58:30,590 - 🚗 TeslaAPI 13 car_api_available returning True
2024-04-24 10:58:30,591 - 🚗 TeslaAPI 13 startOrStop is set to start
2024-04-24 10:58:30,591 - 🚗 TeslaAPI 13 Don't charge Big Red because vehicle.stopAskingToStartCharging == True
2024-04-24 10:58:30,592 - 🚗 TeslaAPI 20 Kriek is not at home.  Do not start charge.
...
2024-04-24 10:58:38,051 - ⛽ Master   20 Charge Session Started for Slave TWC 2119
2024-04-24 10:58:38,060 - ⛽ Slave    20 SHB 2119: 01 01.61/24.00A 0000 0000  M: 00 00.00/00.00A 0000 0000
```

You might also be right about keeping the car awake by polling to see if the `charging_state` changed. 

I'm going to write this off as a fluke between Big Red (Tesla M75D from 2018) and the TWC.

closed 2024-04-29 by RichieB2B