ngardiner/TWCManager #578: Charging at single phase on a 3 phase connection causes incorrect power consumption.

alexeiw123, 2024-04-27:
I have a 2nd EV in the house temporarily that has a single phase charger on board. My TWC is 3 phase and all has been OK charging the tesla.

I noticed today that my house consumption CTs were reporting ~8kW of power while charging this car. TWCmanager however was report in ~24 kW.

Does twcmanager assume the same current on every phase with a voltage present? Does it have any way to measure which phases are actually charging?

RichieB2B, 2024-04-28:
How do you measure your power consumption in TWCManager? What module have you enabled for this?

alexeiw123, 2024-04-28:
> How do you measure your power consumption in TWCManager? What module have you enabled for this?

I'm using MQTT from twcmanager and logging that in home assistant. It also visible on the twcmanager page in realtime, but not logged.

RichieB2B, 2024-04-28:
> > How do you measure your power consumption in TWCManager? What module have you enabled for this?
> 
> I'm using MQTT from twcmanager and logging that in home assistant. It also visible on the twcmanager page in realtime, but not logged.

That sound like you are publishing the TWCManager data to Home Assistant via MQTT. My question is: how are you reading your power consumption in TWCManager? Which EMS module do you have enabled in `/etc/twcmanager/config.json` in the `sources` section? Search for:
```
"enabled": true,
```

alexeiw123, 2024-04-28:
> > > How do you measure your power consumption in TWCManager? What module have you enabled for this?
> > 
> > 
> > I'm using MQTT from twcmanager and logging that in home assistant. It also visible on the twcmanager page in realtime, but not logged.
> 
> That sound like you are publishing the TWCManager data to Home Assistant via MQTT. My question is: how is TWCManager reading your power consumption? Which EMS module do you have enabled in `/etc/twcmanager/config.json` in the `sources` section? Search for:
> 
> ```
> "enabled": true,
> ```

Oh sorry, I use `HASS`

I also have `enphase:` configured to work but disabled it as I've got 1s solar and consumption data coming through HA.

RichieB2B, 2024-04-28:
I understand now. You have a TWC connected to 3-phase power circuit but when you charge using only 1 phase the charging power you see in TWCManager is tripled. 

That is a result of `convertAmpsToWatts()` :

https://github.com/ngardiner/TWCManager/blob/2a6a914657e5414ef96987b1184333c77bc22e75/lib/TWCManager/TWCMaster.py#L245-L247

It gets the amps from `reportedAmpsActual`

https://github.com/ngardiner/TWCManager/blob/2a6a914657e5414ef96987b1184333c77bc22e75/lib/TWCManager/TWCSlave.py#L1178-L1181

Which is the number of amps reported by the TWC in the heartbeat messages. In the car, there is a "3" showing when charging with 3 phases and a "1" when charging with 1 phase. This number is probably somewhere in the heartbeat message as well but that has not been discovered (yet).

So TWCManager doesn't know you are charging with only 1 phase and simply detects that it is connected to 3 phases so it multiplies the report amps by 3.

alexeiw123, 2024-04-28:
> You have a TWC connected to 3-phase power circuit but when you charge using only 1 phase the charging power you see in TWCManager is tripled.

correct!

This is how I noticed it: See the `Export` plot shows I'm exporting while the `EV charger` plot is showing 24 kW (not possible).
(BTW TWC was allowing this because I have it configure for fixed rate between 11am and 2pm due to 0c/kWh at those times).
[image]

> That is a result of `convertAmpsToWatts()` :

That's exactly the piece of math I was hoping to find!
As I suspected, it seems like there's no reasonable way to account for this without having the data available over serial.

I guess it's possible I could create a template entity in HA for my own needs that does some logic between what TWC is reporting and what the house is reporting. or.. I could just ignore the graph as I only have this car (RHD converted F150 lightning) for a few days.

RichieB2B, 2024-04-29:
It's  not too hard to read the 1-phase vs 3-phases usage over MQTT. So if you can find a way to get this signal from Home Assistant we can use it.

MikeBishop, 2024-04-29:
This is one thing that will be improved if we ever manage to merge #483, since I prefer the number of phases reported by the car over that reported by the TWC.

RichieB2B, 2024-04-29:
@MikeBishop I think the problem here is that it not a Tesla charging but another car brand that only has a 1 phase onboard charger. If it was a Tesla it would charge with 3 phases.

alexeiw123, 2024-04-29:
> It's not too hard to read the 1-phase vs 3-phases usage over MQTT. So if you can find a way to get this signal from Home Assistant we can use it.

I can easily get 1s data for each phase of my whole of house production or consumption, in fact it's already coming through in json format over MQTT from my enphase. The issue is that I'm not getting it for the circuit to my TWC, so if TWC is reporting the current/power on each phase then the only option would be to do some kind of fuzzy logic based on the whole of house consumption to work out if the TWC calc of `amps * 1` or `amps * 3` is correct. I can't see that being a practical solution for all users as it would be so specific to each installation.

> This is one thing that will be improved if we ever manage to merge #483, since I prefer the number of phases reported by the car over that reported by the TWC.

That seems like it would work well for so long as you're charging on a tesla with API access. I don't see that it would have the information to show correct power for non tesla cars that take 1 phase of the 3 available.

The current solution is actually quite good in that even a friends tesla will correctly display 1 or 3 phase, depending on how I have the TWC connected. (Yes, I change over my installation between 1 and 3 phase occasionally, and TWC Manager handles that very well when it reboots).

MikeBishop, 2024-05-03:
> @MikeBishop I think the problem here is that it not a Tesla charging but another car brand that only has a 1 phase onboard charger. If it was a Tesla it would charge with 3 phases.

Ah, well. In that case, yes, we have to go with what the TWC is telling us, I think. Since it doesn't expose current indicators for all three phases, I'm not sure that we can differentiate between three-phase and one-phase-out-of-three.

alexeiw123, 2024-05-06:
fair enough. I'll mark this as closed: not planned (read - not possible). Thanks for the discussion!

closed 2024-05-06 by alexeiw123