ngardiner/TWCManager #581: Exception in thread paho-mqtt-client-MQTT.EMS if mqtt

MathildeDec, 2024-05-08:
twcmanager-twcmanager-1  | 21:16:04 🎮 HTTP     20 Serving at port: 8080
twcmanager-twcmanager-1  | Exception in thread paho-mqtt-client-MQTT.EMS:
twcmanager-twcmanager-1  | Traceback (most recent call last):
twcmanager-twcmanager-1  |   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
twcmanager-twcmanager-1  |     self.run()
twcmanager-twcmanager-1  |   File "/usr/lib/python3.9/threading.py", line 892, in run
twcmanager-twcmanager-1  |     self._target(*self._args, **self._kwargs)
twcmanager-twcmanager-1  |   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 4517, in _thread_main
twcmanager-twcmanager-1  |     self.loop_forever(retry_first_connection=True)
twcmanager-twcmanager-1  |   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 2291, in loop_forever
twcmanager-twcmanager-1  | 21:16:04 ⛽ Manager  20 TWC Manager starting as fake Master with id 7777 and sign 77
twcmanager-twcmanager-1  |     rc = self._loop(timeout)
twcmanager-twcmanager-1  |   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 1680, in _loop
twcmanager-twcmanager-1  |     rc = self.loop_read()
twcmanager-twcmanager-1  |   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 2094, in loop_read
twcmanager-twcmanager-1  |     rc = self._packet_read()
twcmanager-twcmanager-1  |   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 3137, in _packet_read
twcmanager-twcmanager-1  |     rc = self._packet_handle()
twcmanager-twcmanager-1  |   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 3811, in _packet_handle
twcmanager-twcmanager-1  |     self._handle_suback()
twcmanager-twcmanager-1  |   File "/usr/local/lib/python3.9/dist-packages/paho/mqtt/client.py", line 4071, in _handle_suback
twcmanager-twcmanager-1  |     on_subscribe(
twcmanager-twcmanager-1  | TypeError: mqttSubscribe() takes from 4 to 5 positional arguments but 6 were given

if MQTT EMS is actif
         "MQTT": {
          "enabled": true,
          "brokerIP": "192.168.xx.xx",
      # The MQTT EMS module subscribes to the below topics to read Consumption/Generation values
          "username": "xxx",
          "password": "xxx",
          "topicConsumption": "zigbee2mqtt/xxxx/energy_produced_b",
          "topicGeneration": "aps2mqtt/aps/xxxx/energy_today"
        },

I think

in https://github.com/ngardiner/TWCManager/tree/main/lib/TWCManager/EMS/MQTT.py
line 109:
    def mqttSubscribe(self, client, userdata, reason_codes, properties=None):
        logger.info("Subscribe operation completed with mid " + str(mid))

would be with mid:

    def mqttSubscribe(self, client, userdata, mid, reason_codes, properties=None):
        logger.info("Subscribe operation completed with mid " + str(mid))

Regard

pflik, 2025-01-19:
yes, also running into this problem.

ngardiner, 2026-04-14:
Thanks for the report. This was fixed in a23aca7 and 71a8e9b