ngardiner/TWCManager #588: Incorporation of Teslemetry Authentication into TWCManager

ChutneyMary, 2024-06-11:
I've started using the attached service for Tesla data (within a Node-Red flow) - https://teslemetry.com/
The only downside is a monthly subscription cost of USD $2.75/month or USD $6.60/3 months.

It'd be great if TWCManager could make use of the Teslemetry authentication routine. Unfortunately, the purchase of a Domain and setting up TWCManager to use a direct procedure is beyond my meagre skills.

Teslemetry also offers a free Webhooks service. This could present TWCManager with a method to reduce API calls (by having a continuous one-way flow of information from vehicle to TWCManager).

danielsan1, 2024-06-15:
Thanks for the finding! It’s the same for me unfortunately - setting up the extra domain/server thing to bring TWCmanager back to normal operations is not working for me :-/ 
@ngardiner @MikeBishop can this provider help  not-so-advanced users to finally make TWCmanager work properly again? (In an easy way)

ngardiner, 2024-06-19:
This has all of the features needed to make TWCManager work (other than actually being part of the codebase today) based on what is published on their website.

It seems a bit expensive for what it is. If someone had a subscription anyway I could see it being of use but if this was just for TWCManager I would not call it good value.

That said, I don't think anything exists today which I would call good value. In theory it would be simple enough to host a web app which did the key generation and exchange with Tesla on your behalf, the problem being trusting that web app since it would have access to your private key.

Then again, for how long have we been using apps/websites which fetched Tesla Auth tokens...

MikeBishop, 2024-06-20:
Trouble is, I'm fairly sure running a service that just resells access to your auth tokens is pretty blatantly against Tesla's terms of use.

Two asides:
- Shameless plug again: You can selfhost the same with https://github.com/MikeBishop/fleetapi-tokens, written by someone who has a fair amount of code in something you're already running, but it requires at least a bit of know-how.  It assumes that you can:
  - register a domain
  - either get a static address and point that domain to it or configure dynamic DNS
  - run a web app in Docker that's accessible to the public on that address
  - register on the Tesla dev site
- While in theory the Tesla Auth tokens apps acted locally on your machine, it's true that any of them _could_ have exfiltrated the token elsewhere as well.  The Fleet API somewhat reduces the scope of such a breach by making refresh tokens single-use -- if the attacker refreshes before you do, your refresh will fail and you'll see the error; if you refresh before the attacker does, they only get access to the car for a few hours after you use the app.

MikeBishop, 2026-10-26:
Closing this. v1.4.0 has a built-in Fleet API login, which covers the main reason for a third-party token service, and there were concerns above about the terms of reselling tokens.

closed 2026-10-26 by MikeBishop