ngardiner/TWCManager #594: Migrate to Open-Source Tokens

MikeBishop, 2024-08-07:
Tesla has just added the option of [Open-Source Tokens](https://developer.tesla.com/docs/fleet-api/authentication/open-source-tokens) to the fleet API, where applications can get client IDs that do _not_ have an associated client secret. That's exactly what we need here, and relieves customers of running their own integration.

The migration looks pretty straightforward and might not even require code changes, but we should a) register TWCManager, b) validate that assumption, and c) update documentation.

MikeBishop, 2024-08-07:
@ngardiner, do you want to do the registration and get TWCManager a token? We shouldn't duplicate.

MikeBishop, 2024-08-07:
Hm, it appears you still have to hard-code URLs. That may be a barrier we need to think through.

dtiefnig, 2024-09-19:
The user auth process looks pretty straight forward and I can't see why it would need to interact with an additional service. Maybe the URLs can be filled with arbitrary values during registration and it would still work? Could easily try with a https://webhook.site/ URL or so and see if anything arrives there.

Found your thread with Tesla btw. and am linking it here as a reference for others.
https://github.com/teslamotors/vehicle-command/issues/111