ngardiner/TWCManager #600: Don't hardcode tesla-control location

MikeBishop, 2025-01-07:
https://github.com/ngardiner/TWCManager/blob/87f9d0ec6af3f5a226490bdcf97353545f3d6981/lib/TWCManager/Vehicle/TeslaBLE.py#L16

Given that not everyone runs with `/home/twcmanager` as their location, I'd suggest not hardcoding this. We don't have any other dependencies on a particular install path that I'm aware of.

The Makefile recipe for tesla-control also fails if that's not the target location, but it is at least responsive to passing a different `HOME` to the make command.

ngardiner, 2025-01-24:
I'd look at it slightly differently (but don't fundamentally disagree with you @MikeBishop) in that it is more a fallback value than hardcoded, as it can be overwritten by any precedence logic we'd like to apply.

I made the initial decision to build the binary within the twcmanager home directly simply because it currently exists as a source distribution and requires building (with a specific golang version no less, it's a bit of a mess if I'm honest), and there's every possibility in the future that distro packaging may be available and we could do it cleanly, but if we build and install to a system path now, we'd end up breaking the packaging later.

An example of how I could improve the logic might be to search for a tesla-control binary within the system path and then fall back to the twcmanager build if one doesn't exist. I do agree that it needs a bit more thought. If it was as easy as installing a package, I'd just do that and rely on system $PATH.

MikeBishop, 2025-01-24:
I was thinking more that not all installations go to that location; I don't have an issue with it looking in the current directory / location of the script for it. I run mine out of `~/TWCManager`, for example, rather than a separate `twcmanager` user's home directory. (After all, it's the only thing running on the device and I'm often running it from the command line for debugging purposes.)

What about allowing the location to be in the config, just like the certificate files?

ngardiner, 2025-01-25:
Good point - I missed your point around specifying an alternate $HOME. I've got an upcoming commit which will cover both of these off.

ngardiner, 2025-01-31:
Closing this off with logic added in 03cd065

closed 2025-01-31 by ngardiner