ngardiner/TWCManager #604: Unable to get energy consumption Tried both Enphase as well as home assistant

pflik, 2025-01-17:
I'm starting to pull my hair out. I have configured both enphase as well as home assistant in the config.json. While loading i'm not getting any errors (loglevel 13) but the consumption is 0. Even though I can clearly see consumption in both enlighten as well as home assistant.

I have not connected twcmanager up to a twc yet. but I don't think that would be the problem?

can anyone guide me a bit in the right direction?

Samiam789, 2025-01-18:
Are you using the enphase module in TWCManager? I think enphase added some sort of extra authentication process for local access. Home assistant was updated to use this but I don't think that the TWCManager enphase module can. I get my enphase solar data via the home assistant module.

pflik, 2025-01-18:
Yeah. I'm also not getting any data from home assistant. The data is in
homeassistant, I can see the meter readings on the dashboard

Samiam789, 2025-01-18:
Is this what you're using to get the solar data from home assistant? https://github.com/ngardiner/TWCManager/blob/main/docs/modules/EMS_HASS.md

pflik, 2025-01-18:
Yes. Filled everything in as needed. It's using https so it's set to true.

But I'm also not getting any errors that the data is not being received.
Something I'm really expecting if no data is coming in

pflik, 2025-01-18:
It's getting weirder. I thought i would just start publishing the sensor results from homeassistant through mqtt. So i configured both the control and sensor MQTT settings. 

I start TWCmanager, i can litterally see in the log `(<mqtterrorcode.mqtt_err_success: 0>, 1)`  which should mean it's connected. Yet when I look in the MQTT tree I see no TWC topic to control or see any values coming from TWCManager (and ofcourse, the sensors values from homeassistant are still not picked up by TWCManager even though i can see them being published on MQTT.

Why does it seem that no data seems to flow (in any direction)

pflik, 2025-01-19:
Just did a fresh reinstall just to make sure that it's not a problem with the install.
 
[image]

in the console i see a steady stream of MQTT messages coming in... but no such thing on the front end.

[image]

i'm not expecting anything on the charging numbers (the TWC is not connected). But consumption generation and solar surplus i do expect to see something. Anyone know what is going on?

On a side note, the car location seems to be off too. I see my car in the dropdown, but when i set the home location it just drops the pin on the northpole... I can assure you that's not my home, nor is the car there. (location 10000, 10000)

RichieB2B, 2025-01-20:
The values published by HASS via MQTT are not being picked up bt the TWCManager MQTT EMS module because of a bug which I fixed in PR #605

Camelroel, 2025-02-03:
A late reply. I had an issue that Home assistant sents out the power data in kW while TWC manager expects the standard watss. So TWC might see hardly any power.

pflik, 2025-02-07:
> A late reply. I had an issue that Home assistant sents out the power data in kW while TWC manager expects the standard watss. So TWC might see hardly any power.

This could be the problem. Is there a way to apply a conversion factor? my p1 meter directly publishes to mqtt and has no option to send in watts.

Camelroel, 2025-02-08:
I don't have experience with mqtt. I use DSMR smart meter to read the values from the P1 port. 

And in the config file from HA I added a section sensor with 2 templates where the value is multiplied by 1000 to get Wats.

The new value names are then put in the TWC manager config file.

I don't know why but it looks like TWC manager wants to see if the solar panels producing energie even if you have a surplus from your P1 meter. 
I struggled with for a while too as I only get data from my solar panels once every 10 minutes. As I have limited polling the Solar Edge system. So I thought to deselect that in TWC but then it wouldn't follow the production. If I give the data from Solar Edge it works and it looks like it uses the values from the p1 port.

RichieB2B, 2025-02-08:
When you have succeeded to configure to send the P1-meter data to TWCManager, there is no need to also send data from the solar production to TWCManager. Simply set this in your `config.json`:
```
    "treatGenerationAsGridDelivery": true,
```
This will make TWCManager try to keep the grid delivery at 0W.

Camelroel, 2025-02-08:
> When you have succeeded to configure HASS to send the P1-meter data to TWCManager, there is no need to also send data from the solar production to TWCManager. Simply set this in your TWCManager `config.json`:
> 
> ```
>     "treatGenerationAsGridDelivery": true,
> ```
> 
> This will make TWCManager try to keep the grid delivery at 0W.

@RichieB2B This is a new line in the config file. As I don't see it in the current config file.

pflik, 2025-02-08:
I will do that. However I do want to have a conversion factor for mqtt
sources. Getting my data from Hass is less then ideal.

It's p1-->mqtt-->hass--> twcmanager

I rather just get the data directly from mqtt

RichieB2B, 2025-02-08:
> -twcmanager-1 | 12:31:13 ⚡ MQTT 18 MQTT EMS Consumption Value updated
> -twcmanager-1 | 12:31:13 ⚡ MQTT 18 MQTT EMS Generation Value updated

These lines mean the MQTT EMS module is working. It receives messages and updates the Comsumption and Generation values. You should see them in the TWCManager dashboard now.

I agree the logging should contain the values to be more useful, hopefully #611 will fix that.

pflik, 2025-02-08:
I think it it may be a conversion issue. Because I don't see anything
beyond 2 watt of usage when actually 2.6 kw is being used