ngardiner/TWCManager #613: TWC 1.3.2, python 3.10 (on armbian pine64) Hass issue [solved with python 3.12], Tesla API ok on 2018 tesla

revoneor, 2025-02-18:
> @revoneor When using `docker-compose` you are running a prebuilt docker image (check your `docker-compose.yml` to see which version) and the python version that comes with it. When using `pyenv` you run TWCManager locally. You cannot combine the two methods. Please open a new issue/discussion. This one is about a specific problem with from a year ago. 

 _Originally posted by @RichieB2B in [#557](https://github.com/ngardiner/TWCManager/issues/557#issuecomment-2662911082)_

Hi Rich
Docker compose now on v1.3.2 
>sudo docker compose -f docker-compose.yml up -d
my .venv lists python 3.12.9
TWC to and from Hass works fine now. 
I retried on my system Python 3.10 (so without .venv) and again Hass failed to report input, so that seems like some strange quirck.

I struggled to get my Tesla API connected, there are so many topics I didn't know what actually is required for a 2018 tesla.
Low and behold after doing that click in the top tesla api banner and posting the error page url into twc it started working.
(I can not re issue the link, the http://registration.teslatasks.com/generateTokens from the manual settings page is different, also generates not found though) (also in the git html I can not find the link yet)
Not sure if that is affected by my tesla fleet certificates hosting and tesla user and dev account with 2FA I was logged in already, I got a client ID and code but have not added that in config.json

Question, is there a way to get the SoC and Target exported?

[image]
My compliments on this great tool to get balanced net using Tesla as storage

[image]
Self use solar power!

RichieB2B, 2025-02-18:
TWCManager currently does not export the SOC of the car and the target for the current charge. It is possible to add this to the [Status MQTT](https://github.com/ngardiner/TWCManager/blob/main/docs/modules/Status_MQTT.md) module if more users require it.

MikeBishop, 2025-02-18:
When you say exported, what are you looking for? Anything visible on the web page is accessible through the REST API. For example, the `getSlaveTWCs` endpoint includes the `lastVIN`, `lastBatterySOC`, and `lastChargeLimit` properties.

revoneor, 2025-02-21:
I guess that takes polling from HASS, I rather see a POST from TWC on status update. But not too important..

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look.

closed 2026-10-26 by MikeBishop