ngardiner/TWCManager #614: Bluetooth pairing - strange behaviour / TWC error log

MiRoCZ, 2025-02-24:
Hello Nathan @ngardiner. 
Any idea what´s going on here with Bluetooth and how to resolve it please ?

[image]

A new (TWC) key was successfully added to my Tesla car through Action button "Pair with Vehicle" (even ..../vehicle/localMgmt TWC web page did not displayed), but BLE Control Status is stil "Not Implemented"

[image]

Can you please share your experience. Any advice really appreciated. Thx.

ngardiner, 2025-02-24:
My apologies for the confusion. It has worked, the Not Implemented is there to reflect that I haven't implemented a way to actively poll a vehicle to give you positive confirmation that it worked, but if the key has paired with the car then it is working.

There is a way for me to confirm that peering has worked (polling vehicles over BLE) so I should have that fixed very shortly, but in the meantime you should be all set for BLE control.

MiRoCZ, 2025-02-26:
Dear Nathan @ngardiner. Thank you very much for your prompt response.

Unfortunatelly BLE control  doesn´t work as there is a issue with Bluetooth on my Rasperry PI 2W - 

[image]

Please see Error: failed to find a BLE device below:

[image]

MiRoCZ, 2025-03-01:
Is there anybody with RPi Zero 2W with build in Bluetooth who has up and running Bluetooth with TWC Manager ?  I´m experiencing Bluetooth down status / busy issue. Any advice really appriciated. 

[image]

@ngardiner  What kind on Bluetooth (RPi) do you use ?

MiRoCZ, 2025-03-01:
The problem is related to the cyclic unsuccessfull initialization of the SAP driver (SAP Bluetooth profile) on RPi Zero 2W build in Bluetooth which results into shut down of the Bluetooth hci0 as is visible on the screenshots below. @ngardiner Does TWC Manager use/need SAP profile initiated ? Does anyone else is experiencing this issue ? What kind of hardware do you use for smooth use with TWC ? I´m really desperate now and don´t know how to make it work.

[image]

hci0 is UP RUNNING shortly after reboot and then goes DOWN when kernel: Bluetooth: hci0: Opcode 0x2010 failed: -16 appear in the log

[image]

ngardiner, 2025-03-02:
> Is there anybody with RPi Zero 2W with build in Bluetooth who has up and running Bluetooth with TWC Manager ? I´m experiencing Bluetooth down status / busy issue. Any advice really appriciated.
> 
> [image]
> 
> [@ngardiner](https://github.com/ngardiner) What kind on Bluetooth (RPi) do you use ?

I'm using an RPi Zero W. Appreciate your troubleshooting of what's causing it @MiRoCZ - We do not have a dependency on SAP profiles, neither do we require bluetoothd to be running. Could you please try:

```systemctl stop bluetoothd```

And see if that helps?

MiRoCZ, 2025-03-02:
Mar 02 11:58:22 TWCManager kernel: Bluetooth: hci0: Opcode 0x c03 failed: -4

MiRoCZ, 2025-03-02:
Hello @ngardiner, TWC rebooted and  systemctl stop bluetooth used, but unfortunatelly no change:

[image]

[image]

ngardiner, 2025-03-02:
Okay, could you try this please:

```systemctl disable bluetooth```

Then reboot, and check again?

> Can you please tell me how long it should take to switch back to use API for stop charging command if Bluetooth not working ? My car is still charging even it should stop.

It should be immediate. The only thing I can think of is that it is detecting that the BLE command is working when it isn't - I'll do some further investigation. There's a PR I am working on which would provide us with real-time stats on whether it's successful or failing which would confirm this for us, I'll see if it's something I can merge early so we can use it for troubleshooting

MiRoCZ, 2025-03-02:
Done a this is a result:

[image]

and after reboot

[image]

MikeBishop, 2026-10-26:
Thanks for raising this. There hasn't been activity here in a long time, and a lot has changed since (the current release is v1.4.0), so I'm closing this out as part of an issue sweep. If you're still running into problems, please re-open this or file a new issue with your version and logs and we'll take a look. Separately, the misleading "Not Implemented" BLE status on the vehicle page is being addressed in the 1.4.1 work.

closed 2026-10-26 by MikeBishop